mathoudebine/turing-smart-screen-python #289: High rate of accessing a background file

sanyer, 2023-07-13:
**Describe the bug**  
Process Monitor shows that this process is accessing background image a lot.
I'm actually not sure that it is accessing the disk each time as the System Informer shows that Disk I/O is just couple of KBs (presumably the size of the image)

**To Reproduce**  
Steps to reproduce the behavior:
1. Run software
2. Check System Informer for the high I/O rate
3. Check Process Monitor for the fact of accessing background file

**Expected behavior**  
Not accessing the file dozens of times per second.

**Screenshots / photos of the Turing screen**  
[image]

**Environment:**  
 - Turing 3.5”
 - 3.0.0
 - Windows 11
 - Python 3.10
 - AMD 5800X3D, NVIDIA GeForce RTX 3070

**Additional context**  
Add any other context about the problem here.

mathoudebine, 2023-08-16:
Hello! Thanks for raising this issue, I agree this behavior is not optimal. It would be better to load the picture only once in memory for the lifetime of the program.
Some themes use different background pictures depending on the elements to display e.g. https://github.com/mathoudebine/turing-smart-screen-python/blob/main/res/themes/Landscape6Grid/theme.yaml uses `background.png` and `background_empty_dark.png`. This should be taken into account when fixing this bug

sanyer, 2023-08-19:
@mathoudebine yeah, I'm using theme Cyberpunk and also see file `generale-mono/GeneraleMonoA.ttf` being accessed around 100 times per second:
[image]

mathoudebine, 2023-09-23:
Hi @sanyer 
The issue should be fixed in the latest commits on `main` branch

closed 2023-09-23 by mathoudebine

sanyer, 2023-09-24:
> Hi @sanyer 
> 
> The issue should be fixed in the latest commits on `main` branch

I will try to take a look tomorrow. Thanks!