mathoudebine/turing-smart-screen-python #305: Frequency Shows as zero (Linux)

wd5gnr, 2023-08-15:
The CPU Frequency was showing as 0.00 GHz no matter what. AMD running recent Linux (NEON/KDE).

The problem is psutils returns a tuple with the speed in GHZ.

The code below was dividing the frequency by 1000 (4/1000 = about 0.00). 

```
Python 3.10.12 (main, Jun 11 2023, 05:26:28) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import psutil
>>> psutil.cpu_freq()
scpufreq(current=4.050250000000001, min=2200.0, max=3800.0)
>>> psutil.cpu_freq().current
4.133458333333334
>>> 
```

This fixes is (in stats.py)

```

    @staticmethod
    def frequency():
        display_themed_value(
            theme_data=config.THEME_DATA['STATS']['CPU']['FREQUENCY']['TEXT'],
            value=f'{sensors.Cpu.frequency():.2f}',
            unit=" GHz",
            min_size=4
        )
```

mathoudebine, 2023-08-16:
Hi, thanks for raising this issue! This is actually a strange behavior from psutil, as you can see the current frequency is given in GHz whereas the min/max is given as MHz.
The expected behavior would be this kind of output:
```
scpufreq(current=2496.0, min=1300.0, max=2496.0)
```
This is a bug from psutil, because the documentation for `cpu_freq()` clearly states: [Return CPU frequency as a named tuple including current, min and max frequencies expressed in Mhz](https://psutil.readthedocs.io/en/latest/index.html?highlight=disk%20usage#psutil.cpu_freq)
It looks like this bug https://github.com/giampaolo/psutil/issues/2049 but it should have been fixed in psutil 5.9.1 used for this project.

Can you try to update psutil to the latest 5.9.5 version to check if it fixed your issue?

wd5gnr, 2023-08-16:
Good call. That does seem to fix it. I should have looked upstream. Thank you!

wd5gnr, 2023-08-16:
Might consider changing requirements.txt to hold back psutil until this gets fixed since it appears it was a regression : psutil~=5.9.5         and that's what I would have installed. I pulled 5.9.1 and that works correctly (I misread your note--I already had 5.9.5 which gave the bad behavior)

mathoudebine, 2023-08-16:
That's very strange because this bug has been fixed in 5.9.1 and the fix is still in the code of the 5.9.5 version. Are you sure you had the 5.9.5 installed when you had the issue? Can you manually force the 5.9.5 installation and check again just to be sure? Thanks

wd5gnr, 2023-08-17:
Well, of course, I can't get it to fail now ;-) I thought I had run through requirements.txt but perhaps I didn't get that one loaded for some reason. Sorry for the false alarm!

closed 2023-09-21 by mathoudebine