mathoudebine/turing-smart-screen-python #309: Adding drop shadows to fonts?

wanderling, 2023-08-23:
**Is your feature request related to a problem? If so, please describe the problem.**  
A description of what the problem is. E.g. I cannot display this data the way I want [...]

**_Text is hard to read on pattered backgrounds._**

**Describe the feature / solution to your problem you'd like**  
A description of what you want to happen.

**_Adding drop shadows or support for chromatic fonts would be a great help._**

**Describe alternatives you've considered / and or tested**  
A description of any alternative solutions or features you've considered.

**Screenshots / photos & mockups of the Turing screen**  
Add screenshots or photos/mockups of the rendering on the Turing screen to help explain your request.
You can drag and drop photos here to add them to the description.

**Environment:**  
 - Revision of this project [e.g. 1.1.1, `main` branch, specific commit]
   3.1.0
 - OS with version [e.g. Windows 11, Ubuntu 22.04]
   Linux Mint 21.2 Victoria 
 - Python version [e.g. Python 3.8]
   Python 3.10.12
 - Hardware [e.g. Intel CPU, Nvidia GPU, Raspberry Pi 3 B+ ...]
   OptiPlex 3020M
   

**Additional context**  
Add any other context or screenshots about the feature request here.

braewoods, 2023-09-22:
This might be doable by overlaying text twice with different colors and slightly offset from each other. This trick can be used to cause text to appear sunken or raised up depending on the colors used.

mathoudebine, 2023-09-25:
Also relevant: draw the text multiple times with 1-pixel offsets: https://itecnote.com/tecnote/python-text-shadow-with-python/