mathoudebine/turing-smart-screen-python #315: Support for 3.5" Screen from KIPYE

braewoods, 2023-09-02:
I found this awhile back and I already have added one to my collection to eventually add support in my own project. It is a 3.5" Screen with no obvious branding, just some writing on the front bezel I cannot decipher. I assume it is Chinese, perhaps the name of the company that I was told to get the control software from.

The product can still be ordered from here: https://www.aliexpress.us/item/3256803899049957.html
The software can be downloaded from here: http://www.kipye.com/en/col.jsp?id=107

Previously I did my own investigation and found the screen also exposes itself as a TTYACM device which means this project should be able to support it. It uses its own custom protocol. I did some digging into the proprietary software and discovered that the entire software is C# and because they shipped a debug build, it is possible to disassemble it into its original source code.

I did so previously to discover what command sequences are available for use. The device appears to use a stateful protocol as it seems to require you to enter a special mode to send image data and a special command to return to the normal command mode. I wasn't able to fully decipher how the orientation code works but it may be possible to do orientation stuff in hardware.

Anyway I thought I would drop this here so it can be considered for addition in the future.

mathoudebine, 2023-09-02:
Hi, nice find ! It would be interesting to add support to this project if possible. If the screen is identified as a serial port it is a good start.
I will order one and add the model to the Readme

braewoods, 2023-09-02:
I can get you some data on the USB device when it is connected to a USB host if you need it for the tables.

mathoudebine, 2023-09-14:
FYI the engraved inscription on the front bezel is 奇叶智显 meaning "Qiye Smart Display" which seems to be the display model name, with Kipye being the manufacturer name. Info have been added to the README and I should receive mine soon!

closed 2023-09-21 by mathoudebine

mathoudebine, 2023-09-21:
Hi @braewoods
Support for Kipye Qiye smart display is now available in the latest release! Feel free to test it and open issues if you see something wrong