mathoudebine/turing-smart-screen-python #335: 3.3.1 - Screen Only Partially Drawing

foxholenoob, 2023-09-24:
**Describe the bug**  
Only two thirds of the LCD screen renders the output.

**To Reproduce**  
1. Using simple-program.py and (Hardware Revision A)
2. Remove the provided sample loop and just do the following:

`background = "res/backgrounds/example_480x320.png"`
`lcd_comm.DisplayBitmap(background)`

**Expected behavior**  
A full screen render. I originally wrote my program with 2.4.0 and the screen rendered fully without issue. I updated to 3.3.1 and the screen is only rendering two thirds. Thinking my LCD might be broken. I restored my old build using 2.4.0 and the screen rendered perfectly still. As far as I can tell going back to 3.1.0 the screen still renders fully.

**Screenshots / photos of the Turing screen**  
[image]

**Environment:**  
 - Smart screen model - Hardware Revision A (Turing Smart Screen 3.5" & UsbPCMonitor 3.5"/5")
 - Revision of this project [3.3.1]
 - OS with version [Windows 11]
 - Python version [Python 3.8]

**Additional context**  
I went through and updated all my package dependencies to match or exceed the requirements.txt

Here is my exact model of LCD screen:
https://www.keebmonkey.com/products/keebmonkey-pc-status-monitor?variant=42057381413079

mathoudebine, 2023-09-24:
Hi, thanks for raising this issue!
I can confirm there has been a regression since the 3.3.0 release for the Turing 3.5 / rev. A display.
I will fix it now

mathoudebine, 2023-09-24:
Issue now fixed in the 3.3.2 release

closed 2023-09-24 by mathoudebine

foxholenoob, 2023-09-24:
Confirmed fixed.

[image]