mathoudebine/turing-smart-screen-python #343: raspberry pi support

evilpete, 2023-10-02:
Is somebody working on raspberry pi support?,   if not, I have a  library that can pull the necessary information (without forking million sub processes per query) if there's interest I could incorporate for raspberry pi support.

closed 2023-10-04 by evilpete

mathoudebine, 2023-10-04:
Hello @evilpete
Is your library source code available somewhere? I'd like to have a look!

evilpete, 2023-10-04:
> Hello @evilpete Is your library source code available somewhere? I'd like to have a look!

I had written the stuff to work with a tiny 3 line Adafruit OLED display  on a rpiZero a few year ago

After messaging you then Reading your code it looks like I "reinvented the wheel" as psutil seem to already does half the stuff I wrote,

the rest of the system and GPU info can be gotten in python with the [gencmd](https://github.com/usalko/gencmd) python lib that communicates directly with the Rpi's GPU with via ["Mailbox" messages](https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface) 

here is demo code salvaged from my rpiZero code:
[rpi_stats.zip](https://github.com/mathoudebine/turing-smart-screen-python/files/12810082/rpi_stats.zip)

Note: I tried to put effort into not relying on calling shell command subprocesses