mathoudebine/turing-smart-screen-python #349: "Ghost" text character even with mono spaced font

Psykotik, 2023-10-03:
**Describe the bug**  
The screen is displaying old character when dynamic text is displayed (for example net stats)

**To Reproduce**  
Steps to reproduce the behavior:
1. Set up a theme with the "DOWNLOADED" statistic
2. Check the unit, it should be "M" for Mega.
3. Download at least 1 giga of data to change the unit
4. Check the unit, it should be only "G" but for some reason it's "GM"
5. Restart your computer / the program to refresh the screen : now it's displaying properly the unit as "G" only

**Expected behavior**  
The "olds" text values should not be displayed anymore, especially with mono spaced font.

**Screenshots / photos of the Turing screen**  

on the first image, check as the "ping" is 3 digits long, and the downloaded statistic is still in the megabytes range. On the second image, one "s" is added to the ping due to it going back to 2 digits and the downloaded unit is GM because it's now in the gigabytes range. 
Note: Ping is a custom sensor using ping3 thanks to the new update. 

[image]

[image]

**Environment:**  
 - Smart screen model : Turing 5"
 - Revision of this project : main
 - OS with version : windows 11
 - Python version : Python 3.11.3
 - Hardware : Intel/Nvidia, on screen

**Additional context**  
I've tested the *align* parameter, as seen [here on this previous issue](https://github.com/mathoudebine/turing-smart-screen-python/issues/171) but the probleme remains the same.
The font used is jetbrains-mono/JetBrainsMono-Bold.ttf, and it works for thnigs like CPU and GPU percentages fine, but not on things with units.

mathoudebine, 2023-10-03:
Thanks for raising this issue, this is a known problem that should have been fixed at least for default sensors. The DOWNLOADED stat may have been missed, I will investigate and fix it.

For your custom sensor, the solution to avoid old character to be displayed after refresh is to define a fixed size for your string. If the value to display has a smaller size some spaces will be added as padding before the value so the old value is always erased.
Check this discussion where I explain it: https://github.com/mathoudebine/turing-smart-screen-python/discussions/110

For your custom sensor with the ping, if it is correct to say the ping will be 4-digits max (9999ms) then you could format the value as follow:
```python
return f"{self.as_numeric():>4}ms"
```
This way if the ping is 23ms the text sent to the screen will be `  23ms` with two extra spaces in case the previous value was `1250ms`

Psykotik, 2023-10-03:
Thanks, that a pretty neat way to handle the situation, haven't thought about that. 
I missed the issue #110 even if I tried to check every one. 

Good work thanks !

closed 2023-10-03 by Psykotik

mathoudebine, 2023-10-03:
Thanks, I added this info in a comment on `sensors_custom.py` and also in the Troubleshooting page for other to see: https://github.com/mathoudebine/turing-smart-screen-python/wiki/Troubleshooting#all-platforms-part-of-old-value-still-displayed-after-refresh-ghosting
Have fun!