mathoudebine/turing-smart-screen-python #354: Can't get python dependencies installed

Paradoxum, 2023-10-05:
[image]

**To Reproduce**  
Steps to reproduce the behavior:
run > python.exe -m pip install -r requirements.txt

**Expected behavior**  
installs fine

**Environment:**  
 - SXuanFang 3.5
 - latest
 - w11 22h2 22621.2361
 - python-3.12.0-amd64.exe

Paradoxum, 2023-10-05:
Forgot to add bottom of shell;' 
[image]

mathoudebine, 2023-10-06:
Hello, Python 3.12 is not compatible with this project for now.
This is because we use some libraries that have not been updated yet.

For now you need to keep Python 3.11 until this issue is fixed. Don't worry, Python 3.11 is still actively supported until 2027-10!
You can uninstall Python 3.12 and install Python 3.11 from here: [python.org/downloads/release/python-3116](https://www.python.org/downloads/release/python-3116/) downloads are at the end of the page

mathoudebine, 2023-11-05:
Hello again, Python 3.12 support is now enabled on the main branch (and in the future release)
Of course you can keep Python 3.11 that will work fine as well

closed 2023-11-05 by mathoudebine