mathoudebine/turing-smart-screen-python #358: Can't run configure.py

pingyaru, 2023-10-12:
**Problem**  
I was trying to make step 4 "Configure system monitor" in the "how to start" guide, but it won't start. I was using `C:\Users\Sergey\AppData\Local\Programs\Python\Python311\python.exe "D:\small programs\turing-smart-screen-python-3.3.3\configure.py"` command to run configure.py. I also tried to run cmd as and administrator. 

[image]

`C:\Windows\System32>C:\Users\Sergey\AppData\Local\Programs\Python\Python311\python.exe "D:\small programs\turing-smart-screen-python-3.3.3\configure.py"
Traceback (most recent call last):
  File "D:\small programs\turing-smart-screen-python-3.3.3\configure.py", line 434, in <module>
    configurator = TuringConfigWindow()
                   ^^^^^^^^^^^^^^^^^^^^
  File "D:\small programs\turing-smart-screen-python-3.3.3\configure.py", line 148, in __init__
    self.window.iconphoto(True, PhotoImage(file="res/icons/monitor-icon-17865/64.png"))
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Sergey\AppData\Local\Programs\Python\Python311\Lib\tkinter\__init__.py", line 4125, in __init__
    Image.__init__(self, 'photo', name, cnf, master, **kw)
  File "C:\Users\Sergey\AppData\Local\Programs\Python\Python311\Lib\tkinter\__init__.py", line 4072, in __init__
    self.tk.call(('image', 'create', imgtype, name,) + options)
_tkinter.TclError: couldn't open "res/icons/monitor-icon-17865/64.png": no such file or directory`

I made path to 64.png absolute in configure.py but next it said there was a problem with path to config.yaml. I cound't solve it, so I decided to open an issue

[image]

**Environment:**  
 - Smart screen model: XuanFang 3.5" Revision flagship
 - OS with version: Windows 11
 - Python version: Python 3.11
 - Hardware: gigabyte b760 ds3h ddr4, i5-12400f, sapphire rx 6700 xt, Adata XPG gammix D20 32gb, Kingston KC3000 1tb

mathoudebine, 2023-10-17:
Hi, there is indeed an issue with `main.py` and `configure.py`. They have been written to be launched from their own directory, because they search for resources in the current terminal folder.

Until it is fixed, you need first to move inside your Terminal to the folder where you have downloaded the program:
``` cmd
D:
cd "D:\small programs\turing-smart-screen-python-3.3.3\"
```
then you can run the `main.py` or `configure.py` from there:
``` cmd
C:\Users\Sergey\AppData\Local\Programs\Python\Python311\python.exe configure.py
C:\Users\Sergey\AppData\Local\Programs\Python\Python311\python.exe main.py
```

pingyaru, 2023-10-17:
> Hi, there is indeed an issue with `main.py` and `configure.py`. They have been written to be launched from their own directory, because they search for resources in the current terminal folder.
> 
> Until it is fixed, you need first to move inside your Terminal to the folder where you have downloaded the program:
> 
> ```batchfile
> D:
> cd "D:\small programs\turing-smart-screen-python-3.3.3\"
> ```
> 
> then you can run the `main.py` or `configure.py` from there:
> 
> ```batchfile
> C:\Users\Sergey\AppData\Local\Programs\Python\Python311\python.exe configure.py
> C:\Users\Sergey\AppData\Local\Programs\Python\Python311\python.exe main.py
> ```

It worked! Thank you very much!

closed 2024-12-22 by mathoudebine