mathoudebine/turing-smart-screen-python #359: Linux RPI systemd autostartup not working

SeimusS, 2023-10-14:
Hello,

I am using RPI 4B, I am able to run the script manualy however the moment I try to set ti to run automaticaly it wont work

pi@retropie:~ $ systemctl status turing-smart-screen-python.service
● turing-smart-screen-python.service - Turing Smart Screen
   Loaded: loaded (/etc/systemd/system/turing-smart-screen-python.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2023-10-14 21:46:45 CEST; 7s ago
  Process: 19310 ExecStart=/usr/local/bin/python3.8 /home/pi/turing-smart-screen-python/main.py (code=exited, status=1/FAILURE)
 Main PID: 19310 (code=exited, status=1/FAILURE)

Oct 14 21:46:45 retropie systemd[1]: turing-smart-screen-python.service: Service RestartSec=100ms expired, scheduling restart.
Oct 14 21:46:45 retropie systemd[1]: turing-smart-screen-python.service: Scheduled restart job, restart counter is at 5.
Oct 14 21:46:45 retropie systemd[1]: Stopped Turing Smart Screen.
Oct 14 21:46:45 retropie systemd[1]: turing-smart-screen-python.service: Start request repeated too quickly.
Oct 14 21:46:45 retropie systemd[1]: turing-smart-screen-python.service: Failed with result 'exit-code'.
Oct 14 21:46:45 retropie systemd[1]: Failed to start Turing Smart Screen.

pi@retropie:~ $ journalctl _PID=19310
-- Logs begin at Sat 2023-10-14 19:23:24 CEST, end at Sat 2023-10-14 21:47:01 CEST. --
Oct 14 21:46:44 retropie python3.8[19310]: Traceback (most recent call last):
Oct 14 21:46:44 retropie python3.8[19310]:   File "/home/pi/turing-smart-screen-python/main.py", line 63, in <module>
Oct 14 21:46:44 retropie python3.8[19310]:     from library.log import logger
Oct 14 21:46:44 retropie python3.8[19310]:   File "/home/pi/turing-smart-screen-python/library/log.py", line 30, in <module>
Oct 14 21:46:44 retropie python3.8[19310]:     RotatingFileHandler("log.log", maxBytes=1000000, backupCount=0),  # Log in textfile max 1MB
Oct 14 21:46:44 retropie python3.8[19310]:   File "/usr/local/lib/python3.8/logging/handlers.py", line 148, in __init__
Oct 14 21:46:44 retropie python3.8[19310]:     BaseRotatingHandler.__init__(self, filename, mode, encoding, delay)
Oct 14 21:46:44 retropie python3.8[19310]:   File "/usr/local/lib/python3.8/logging/handlers.py", line 55, in __init__
Oct 14 21:46:44 retropie python3.8[19310]:     logging.FileHandler.__init__(self, filename, mode, encoding, delay)
Oct 14 21:46:44 retropie python3.8[19310]:   File "/usr/local/lib/python3.8/logging/__init__.py", line 1143, in __init__
Oct 14 21:46:44 retropie python3.8[19310]:     StreamHandler.__init__(self, self._open())
Oct 14 21:46:44 retropie python3.8[19310]:   File "/usr/local/lib/python3.8/logging/__init__.py", line 1172, in _open
Oct 14 21:46:44 retropie python3.8[19310]:     return open(self.baseFilename, self.mode, encoding=self.encoding)
Oct 14 21:46:44 retropie python3.8[19310]: OSError: [Errno 30] Read-only file system: '/home/pi/turing-smart-screen-python/log.log'
pi@retropie:~ $

Can somebody advice? I tried to set privilages to the folder, log.log and other stuff nothing makes it to run.

SeimusS, 2024-01-30:
Hello @mathoudebine 

Thank you for the reply. No worries its late, better then never I say :)

I checked the other thread and the Tshoot guide as well the updated Wiki. Looks like this could be the issue as I have the folder in my home thus as you mentioned in the documentation looks like really permission problem due to the fact its in home as well it has additional protection parameters.

Currently as a workaround e.g another possibility to run it. I did create a small bash script that is invoked during reboot in CRON.

```
#! /bin/bash

cd /home/pi/turing-smart-screen-python 
nohup python3 main.py &

```
`@reboot sudo /home/pi/turing-smart-screen-python/start-up-script.sh`

This is working for me as well pretty much without problem. Your solution to run this upon start is much more sophisticated but this is a simple way too.

I think we can close this topic, as I have a working solution and it looks like the issue is indeed caused by wrong location of the folder in home instead opt or by disabling the additional protection parameters in the systemd script.

Regards,
S.

mathoudebine, 2024-02-19:
Thanks for your feedback, glad to hear you find a workaround for this!

closed 2024-02-19 by mathoudebine