mathoudebine/turing-smart-screen-python #362: Erro de conexao

Digaruti01, 2023-10-19:
I recently bought the 3.5 screen, but it is fixed to a white screen and I don't know what to do to make it work as I expected. I tried several times to install and uninstall the app, but it remains on the white screen, I'm about to return it to the seller.
[image]

FabioZumbi12, 2023-10-19:
Try to use another usb cable. I got a similar issue when using an incompatible usb cable (dont know why). Check the revision on config too.

mathoudebine, 2023-10-20:
Can you show us what you have under "Serial ports" in your Device Manager? 
[image]

mathoudebine, 2024-01-29:
Closing this issue since inactive, please tag me if you want to reopen it.

closed 2024-01-29 by mathoudebine