mathoudebine/turing-smart-screen-python #364: Wrong CPU usage

FabioZumbi12, 2023-10-19:
**Describe the bug**  
I having some wrong CPU usages, most in games.

**To Reproduce**  
Steps to reproduce the behavior:
Open a game, open the taskmanager, compare two CPU usages. One shows +-50%, and the display shows +-3%.

**Environment:**  
 - Smart screen model: Hardware Revision A (Turing Smart Screen 3.5")
 - Revision of this project: main
 - OS with version: Windows 11
 - Python version: 3.11.5
 - Hardware AMD Ryzen 5800/RTX 3060

On display:
[image]
On Taskmanager: 
[image]
Note that  the memory it matches exact, only CPU its wrong.

FabioZumbi12, 2023-10-19:
So to known, using `HW_SENSORS: LHM` show a more precise CPU usage with a difference of +-4% between screen and taskmanager

mathoudebine, 2023-10-20:
Thanks for investigating, I knew the Python sensor reading was less precise than LHM but it should not be that far from the "real" data. I will have a look into it and try to understand. Though it is probably an issue with the `psutil` Python package we get the data from.