mathoudebine/turing-smart-screen-python #382: CPU temperature not supported - AMD FX-8350

Parnassous, 2023-11-02:
Hi , 

The CPU temperature is not showing for my CPU AMD FX-8350 
The GPU temperature is working normal

Turing Release: 3.3.3
LibreHardwareMonitorLib: 0.9.2.0
Turing Smart Screen 3.5"
Windows OS

I have the bellow message:
[WARNING] Your CPU temperature is not supported yet

Is there an workaround to get CPU temperature working?

Thanks.

braewoods, 2023-11-05:
This is really an issue for the [upstream project](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues). Try using their standalone program and see if it works there or not.

mathoudebine, 2023-11-05:
@Parnassous The standalone LibreHardwareMonitor can be downloaded here: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/releases/download/v0.9.2/LibreHardwareMonitor-net472.zip
Can you download and extract it on your computer, then eun `LibreHardwareMonitor.exe`and see if your CPU temperatures are displayed in the program?

If yes, then please run this Python program (from this project this time):
`turing-smart-screen-python/external/LibreHardwareMonitor/test_librehardwaremonitor.py`
Then copy the output and paste it here.

mathoudebine, 2023-12-02:
Hello @Parnassous did you manage to solve this issue?

mathoudebine, 2023-12-22:
Closed for inactivity, please comment and tag me if you have new info

closed 2023-12-22 by mathoudebine