mathoudebine/turing-smart-screen-python #389: windows start up error

Ntonik7, 2023-11-08:
**Describe the bug**  
I have create Task Scheduler for script and does not run when i log on windows
When i run it manual the script run without problem

**Environment:**  
 - Turing Release: 3.3.3
 - Smart screen model [Turing Smart Screen 3.5]
 - LibreHardwareMonitorLib 0.9.2.0
 - OS with version [Windows 11 , Windows 10]
 - Python version [Python 3.12]
 - Hardware [Intel CPU 10400F, Nvidia GPU 3060Ti]

**Additional context**  
Install path for python/pythonw and turing script is C:\ 

Thanks for your time

mathoudebine, 2023-11-09:
Hello, can you check your scheduled task from the task scheduler? It should be setup like this:
[image]
[image]
[image]

Can you also send the content of your `log.log` file that should be on the main folder?

Ntonik7, 2023-11-09:
Hi,
Thanks for your answer

I attach my settings, is in Greek Language(sorry about that) and time zone from log if you need is GMT+2

[image]
[image]
[image]

**logs when i run it manually**
[image]

When i reboot my pc and script dont run from scheduled task log file has not added anything new

**Logs from scheduled task**
[image]

I hope this helps and again thanks for your time

Inquisitor956, 2024-05-21:
Hello, everyone!

This issue of Task Scheduler not launching the Python script at logon has frustrated me for a few hours. I followed every step to the letter and nothing seemed to work.

I did find a workaround, however. It involves pointing Task Scheduler to run a batch file at logon instead. I went this route since I could manually launch my configured script just fine using cmd or PS. This batch file contains a PowerShell command that uses pythonw to run the script. Its very simple and it only involved four lines. The batch file points PowerShell to the directory where you have the python script, then runs it using pythonw, and closes the window automatically after it is done. The window disappears in an instant at logon, so its not intrusive at all IMO.

You can copy and paste this into a `.txt` file, edit the second line with the _directory_ where you have your script, then save it as a `.bat file. `

> @echo off
cd C:\insert_turing_smart_screen_directory_here (ex. C:\turing-smart-screen-python-3.4.0)
powershell -NoProfile -Command "Start-Process pythonw.exe -ArgumentList 'main.py' -NoNewWindow"
exit

Configure task scheduler as the wiki instructs, except you select the batch file as the 'program' instead of pythonw.exe. Leave the "Add arguments" and "Run in" fields blank, as the batch file takes care of those. Make sure to still select "Run with highest privileges" or else it will not work! See attached screenshots as examples.

[image]
[image]
[image]
[image]
[image]
[image]

mathoudebine, 2025-06-17:
Closing issue due to inactivity