mathoudebine/turing-smart-screen-python #392: test_librehardwaremonitor.py does not show detailed Motherboard info - the standalone .exe does provide this

exelanz, 2023-11-13:
**Describe the bug**  
When I run LibreHardwareMonitor.exe I get detailed info on my Motherboard. When using the test_librehardwaremonitor.py I don´t get this information.

**To Reproduce**  
Steps to reproduce the behavior:
1. Run LibreHardwareMonitor.exe
2. Run test_librehardwaremonitor.py
3. Compare output

**Expected behavior**  
Same information from both apps

**Screenshots / photos of the Turing screen**  
Output of LibreHardwareMonitor.exe:
[image]

Output of test_librehardwaremonitor.py (Just name of the motherboard, then it continues with storage components):
[image]

**Environment:**  
 - Smart screen model: XuanFang 3.5”
 - Revision of this project: latest release
 - OS with version: Windows 11
 - Python version: tested with 3.11 and 3.12
 - Hardware:  ASUS TUF GAMING B550 plus WiFi II, AMD Ryzen 5600x and AMD Radeon rx 6700 xt
 
**Additional context**  
None - but can provide additional info if needed

mathoudebine, 2023-11-19:
Hello! Thanks for reporting this issue, I can confirm it because I have the same motherboard as you, which will be easier to debug!

mathoudebine, 2023-11-19:
I can confirm the behavior was linked to the SubHardware not being displayed. This branch https://github.com/mathoudebine/turing-smart-screen-python/tree/fix/392-test_librehardwaremonitorpy-does-not-show-detailed-motherboard-info-the-standalone-exe-does-provide-this now contains a fix to list these SubHardware in `test_librehardwaremonitor.py`, can you check if it fixes the issue for you?

exelanz, 2023-11-20:
Yes the fix in the specified branch works for me. Output is as expected. Thanks!

closed 2023-11-20 by mathoudebine