mathoudebine/turing-smart-screen-python #400: Theme Editor not usable on HiDPI

orzel, 2023-11-27:
On a HiDPI display, the theme editor looks like this. When clicking, coordinates aren't visible. And it would tremendously help if the preview could be scaled as well (by the current display scale factore, most of the time it's '2' / 200%).

The config GUI also display bad and is barely usable (you can't read buttons text for exemple), but it's less an issue i guess.

[image]

What's not clear on the screenshot is that the whole window is extremely small.

mathoudebine, 2023-12-02:
Thanks for raising this issue.
Can you tell me more about your OS and how it is configured regarding the scale factor?
I tried to reproduce the behavior on Windows with a 200% scale factor but the GUIs render fine. It seems the complete window is scaled by the OS (including its size) and not each elements like in your preview

orzel, 2023-12-04:
Sorry, I should have said : this is using linux/kde/plasma/x11 (not wayland). The distro is gentoo, meaning very recent software (last releases, last bugs). You can probably reproduce using virtualbox + any x11-based distribution.  Not sure how to have a 'hidpi' virtualbox.
I can do some tests here if needed.

jonathanhirschman, 2024-08-11:
I have the same issue. I'm on Ubuntu 24.04, running Gnome, 4k display. Same issue on iether x11 or Wayland. Desktop is scaled to 200%. UI is very small, and doesn't scale correctly - you can get a sense of how small the ui is from the size of the title bars.

[image]
[image]