mathoudebine/turing-smart-screen-python #401: Theme Editor crashes when using vim

orzel, 2023-11-27:
When using `vim` and saving the file, the python code freaks out and the theme editor window dissapears.

```
27/11/2023 14:13:34 [DEBUG] The theme file has been updated, the preview window will refresh
Traceback (most recent call last):
  File "/home/orzel/clones/turing-smart-screen-python/theme-editor.py", line 263, in <module>
    if os.path.getmtime(theme_file) > last_edit_time:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen genericpath>", line 55, in getmtime
FileNotFoundError: [Errno 2] File o directory non esistente: 'res/themes/test/theme.yaml'
```

I'm not sure about the details, but I think `vim` handles filesaving differently than other editors. It uses _rename_ instead of overwritting.

I use the following change, which works, but might be considered a _hack_, or too _quick & dirty_, I'll let you decide:

`        if os.path.exists(theme_file) and os.path.getmtime(theme_file) > last_edit_time:`

mathoudebine, 2023-12-02:
Hi @orzel, thnaks for raising this issue!
I'm okay with your change, it cannot do harm to check if the file exists in any case.
Do you want to create a PR for this? Otherwise I can commit the change myself.

orzel, 2023-12-04:
For this one-liner, that's probably easier for you to just commit that.

closed 2023-12-22 by mathoudebine