mathoudebine/turing-smart-screen-python #405: 3.5" Screen Won't Work

Paradoxum, 2023-12-02:
**Describe the bug**  
When starting it, it stops and does not continue.

**To Reproduce**  
Steps to reproduce the behavior:
If I start the configurator and select the right everything including COM port (checking with device manager), then save/start, this is the log:
02/12/2023 01:32:19 AM [INFO] Loading theme Terminal from res/themes/Terminal/theme.yaml
02/12/2023 01:32:23 AM [DEBUG] HW revision: B
02/12/2023 01:32:23 AM [DEBUG] Static COM port: COM10
02/12/2023 01:32:24 AM [DEBUG] Found LibreHardwareMonitorLib 0.9.2.0
02/12/2023 01:32:29 AM [INFO] Found CPU: Intel Core i7-7700HQ
02/12/2023 01:32:29 AM [INFO] Found Memory: Generic Memory
02/12/2023 01:32:29 AM [INFO] Found Nvidia GPU: NVIDIA GeForce GTX 1060 with Max-Q Design
02/12/2023 01:32:29 AM [INFO] Found Intel GPU: Intel(R) HD Graphics 630
02/12/2023 01:32:29 AM [INFO] Found Storage: Fanxiang S501 256GB
02/12/2023 01:32:29 AM [INFO] Found Storage: CT2000MX500SSD1
02/12/2023 01:32:29 AM [INFO] Found Network interface: Bluetooth Network Connection
02/12/2023 01:32:29 AM [INFO] Found Network interface: Ethernet
02/12/2023 01:32:29 AM [INFO] Found Network interface: Ethernet 2
02/12/2023 01:32:29 AM [INFO] Found Network interface: Local Area Connection
02/12/2023 01:32:29 AM [INFO] Found Network interface: Local Area Connection* 1
02/12/2023 01:32:29 AM [INFO] Found Network interface: Local Area Connection* 10
02/12/2023 01:32:29 AM [INFO] Found Network interface: WiFi
02/12/2023 01:32:29 AM [DEBUG] Using Python 3.12.0 (tags/v3.12.0:0fb18b0, Oct  2 2023, 13:03:39) [MSC v.1935 64 bit (AMD64)]
02/12/2023 01:32:29 AM [INFO] Tray icon has been displayed
02/12/2023 01:32:30 AM [DEBUG] HW sub-revision: 2578
02/12/2023 01:32:30 AM [DEBUG] Drawing Image: BACKGROUND
02/12/2023 01:32:30 AM [DEBUG] Bitmap res/themes/Terminal/background.png is now loaded in the cache
02/12/2023 01:32:30 AM [DEBUG] Drawing Text: TEXT_EXAMPLE

cmd window closes here.

**Screenshots / photos of the Turing screen**  
Add screenshots or photos of the rendering on the smart screen to help explain your problem.
You can drag and drop photos here to add them to the description.

**Environment:**  
 - Smart screen model [e.g. Turing 5", XuanFang 3.5”]
 - Revision of this project [e.g. 3.0.0, `main` branch, specific commit]
 - OS with version [e.g. Windows 11, Ubuntu 22.04]
 - Python version [e.g. Python 3.10]
 - Hardware [e.g. Intel CPU, Nvidia GPU, Raspberry Pi 3 B+ ...]

**Additional context**  
Add any other context about the problem here.
This is the log of running the 5" screen with it;
02/12/2023 01:42:06 AM [INFO] Loading theme NZXT_C from res/themes/NZXT_C/theme.yaml
02/12/2023 01:42:06 AM [DEBUG] HW revision: C
02/12/2023 01:42:06 AM [DEBUG] Static COM port: COM6
02/12/2023 01:42:07 AM [DEBUG] Found LibreHardwareMonitorLib 0.9.2.0
02/12/2023 01:42:09 AM [INFO] Found CPU: Intel Core i7-7700HQ
02/12/2023 01:42:09 AM [INFO] Found Memory: Generic Memory
02/12/2023 01:42:09 AM [INFO] Found Nvidia GPU: NVIDIA GeForce GTX 1060 with Max-Q Design
02/12/2023 01:42:09 AM [INFO] Found Intel GPU: Intel(R) HD Graphics 630
02/12/2023 01:42:09 AM [INFO] Found Storage: Fanxiang S501 256GB
02/12/2023 01:42:09 AM [INFO] Found Storage: CT2000MX500SSD1
02/12/2023 01:42:09 AM [INFO] Found Network interface: Bluetooth Network Connection
02/12/2023 01:42:09 AM [INFO] Found Network interface: Ethernet
02/12/2023 01:42:09 AM [INFO] Found Network interface: Ethernet 2
02/12/2023 01:42:09 AM [INFO] Found Network interface: Local Area Connection
02/12/2023 01:42:09 AM [INFO] Found Network interface: Local Area Connection* 1
02/12/2023 01:42:09 AM [INFO] Found Network interface: Local Area Connection* 10
02/12/2023 01:42:09 AM [INFO] Found Network interface: WiFi
02/12/2023 01:42:09 AM [DEBUG] Using Python 3.12.0 (tags/v3.12.0:0fb18b0, Oct  2 2023, 13:03:39) [MSC v.1935 64 bit (AMD64)]
02/12/2023 01:42:09 AM [INFO] Tray icon has been displayed
02/12/2023 01:42:09 AM [INFO] Display reset (COM port may change)...
02/12/2023 01:42:24 AM [DEBUG] Static COM port: COM6
02/12/2023 01:42:25 AM [WARNING] Display returned unknown sub-revision on Hello answer (b'')
02/12/2023 01:42:25 AM [DEBUG] HW sub-revision: SubRevision.UNKNOWN
02/12/2023 01:42:25 AM [INFO] Calling ScreenOn
02/12/2023 01:42:26 AM [DEBUG] Drawing Image: BACKGROUND
02/12/2023 01:42:26 AM [DEBUG] Bitmap res/themes/NZXT_C/background.png is now loaded in the cache
02/12/2023 01:42:27 AM [DEBUG] Drawing Text: CPU_MODEL
02/12/2023 01:42:27 AM [DEBUG] Drawing Text: GPU_MODEL
02/12/2023 01:42:28 AM [WARNING] Found 2 GPUs on your system (0 AMD / 1 Nvidia / 1 Intel). Auto identify which GPU to use.

then that closes also.
The 5" one works with the official software fine, the 3" one does not work with this new version I'm seeing but I do remember it did with the older one, now it just says can't find device.

config.yaml

config:
  # Configuration values to set up basic communication
  # Set your COM port e.g. COM3 for Windows, /dev/ttyACM0 for Linux...
  # Use AUTO for COM port auto-discovery (may not work on every setup)
  # COM_PORT: "/dev/ttyACM0"
  # COM_PORT: "COM3"
  COM_PORT: COM10

  # Theme to use (located in res/themes)
  # Use the name of the folder as value
  THEME: Terminal

  # Hardware sensors reading
  # Choose the appropriate method for reading your hardware sensors:
  # - PYTHON         use Python libraries (psutils, GPUtil...) to read hardware sensors (supports all OS but not all HW)
  # - LHM            use LibreHardwareMonitor library to read hardware sensors (Windows only - NEEDS ADMIN RIGHTS)
  # - STUB / STATIC  use random/static data instead of real hardware sensors
  # - AUTO           use the best method based on your OS: Windows OS will use LHM, other OS will use Python libraries
  HW_SENSORS: AUTO

  # Network interfaces
  # Linux/MacOS interfaces are named "eth0", "wlan0", "wlp1s0", "enp2s0"...
  # For Windows use the interfaces pretty name: "Ethernet 2", "Wi-Fi", ...
  # Leave the fields empty if the card does not exist on your setup
  ETH: Local Area Connection # Ethernet Card
  WLO: WiFi # Wi-Fi Card

display:
  # Display revision:
  # - A      for Turing 3.5" and UsbPCMonitor 3.5"/5"
  # - B      for Xuanfang 3.5" (inc. flagship)
  # - C      for Turing 5"
  # - D      for Kipye Qiye Smart Display 3.5"
  # - SIMU   for 3.5" simulated LCD (image written in screencap.png)
  # - SIMU5  for 5" simulated LCD
  # To identify your smart screen: https://github.com/mathoudebine/turing-smart-screen-python/wiki/Hardware-revisions
  REVISION: B

  # Display Brightness
  # Set this as the desired %, 0 being completely dark and 100 being max brightness
  # Warning: revision A display can get hot at high brightness!
  BRIGHTNESS: 100

  # Display reverse: true/false
  # Set to true to reverse display orientation (landscape <-> reverse landscape, portrait <-> reverse portrait)
  # Note: Display basic orientation (portrait or landscape) is defined by the theme you have selected
  DISPLAY_REVERSE: false

from a clean zip. it used to work but I haven't had my PC running for a few months so just playing with em on my laptop.

https://imgur.com/a/yI1XLZi

mathoudebine, 2023-12-02:
First things I can think of:
- check that your COM port COM10 is the correct one for your display
- check that your 3.5" display is a revision B (XuanFang) and not revision A (Turing). To identify your smart screen: https://github.com/mathoudebine/turing-smart-screen-python/wiki/Hardware-revisions
- check all required packages from `requirements.txt` are installed fine
- check the original Turing / XuanFang software is not running in the background

Paradoxum, 2023-12-03:
>check that your COM port COM10 is the correct one for your display
Yes I did that by watching device manager and pulling/plugging the USB cable.

I have the revision flagship. I have previously had this screen working a few months ago.

Original software not running and was not even used on this windows install until now, but I dunno what they did cause that won't work either "not connected".

I'll run the requirements command again.
Check it out;
https://streamable.com/y4x1qj

The 5" one works fine in the turing software.

mathoudebine, 2024-01-10:
If the original software is not working, then maybe your screen is not working anymore?
You can try to download it again from the manufacturer link in case yours is broken: https://github.com/mathoudebine/turing-smart-screen-python/wiki/Vendor-apps#xuanfang-screen-version-b--flagship---vendor-app

closed 2024-01-14 by Paradoxum