mathoudebine/turing-smart-screen-python #415: Can't run from linux, got "Read-only file system" for logger

vlad-berdinskiy, 2023-12-15:
**Describe the bug**  
Hi! I'm used your app for a long time but few month ago it stop working. I reinstalled it and since then can't make it works. Systemctl task  can't start, in logs I see this:

```
Started Turing Smart Screen.
 python3[10610]: Traceback (most recent call last):
 python3[10610]:   File "/home/vlad/Downloads/turing-smart-screen-python/main.py", line 63, in <module>
 python3[10610]:     from library.log import logger
 python3[10610]:   File "/home/vlad/Downloads/turing-smart-screen-python/library/log.py", line 30, in <module>
 python3[10610]:     RotatingFileHandler("log.log", maxBytes=1000000, backupCount=0),  # Log in textfile max 1MB
 python3[10610]:   File "/usr/lib/python3.10/logging/handlers.py", line 155, in __init__
 python3[10610]:     BaseRotatingHandler.__init__(self, filename, mode, encoding=encoding,
 python3[10610]:   File "/usr/lib/python3.10/logging/handlers.py", line 58, in __init__
 python3[10610]:     logging.FileHandler.__init__(self, filename, mode=mode, 
 python3[10610]:   File "/usr/lib/python3.10/logging/__init__.py", line 1169, in __init__
 python3[10610]:     StreamHandler.__init__(self, self._open())
 python3[10610]:   File "/usr/lib/python3.10/logging/__init__.py", line 1201, in _open
 python3[10610]:     return open_func(self.baseFilename, self.mode,
 python3[10610]: OSError: [Errno 30] Read-only file system: '/home/vlad/Downloads/turing-smart-screen-python/log.log'
 systemd[1]: turing-smart-screen-python.service: Main process exited, code=exited, status=1/FAILURE
 systemd[1]: turing-smart-screen-python.service: Failed with result 'exit-code'.
 systemd[1]: turing-smart-screen-python.service: Scheduled restart job, restart counter is at 1.
 systemd[1]: Stopped Turing Smart Screen.
```

**To Reproduce**  
Steps to reproduce the behavior:
1. Use tutorial for Ubuntu
2. Restart system
3. Check journalctl

**Environment:**  
 - Smart screen model - Turing 5"
 - Revision of this project - `main` branch, 0f68e15b024a5099e57ace3f969c1219e467df08
 - OS with version - Ubuntu 22.04.3
 - Python version - Python 3.11.5

mathoudebine, 2023-12-22:
Hi @vlad-berdinskiy 
It seems to be some permission issue on your Linux installation
Can you run
```
whoami
ls -la /home/vlad/Downloads/turing-smart-screen-python
```
and post the output here please?

vlad-berdinskiy, 2024-01-05:
Sure, here it is
```
➜  ~ whoami                                                
vlad
➜  ~ ls -la /home/vlad/Downloads/turing-smart-screen-python

total 2748
drwxrwxrwx  8 vlad vlad    4096 Dec 15 13:09 .
drwxr-xr-x 14 vlad vlad    4096 Jan  4 15:57 ..
-rwxrwxrwx  1 vlad vlad     546 Nov 22 20:47 AUTHORS
-rwxrwxrwx  1 vlad vlad    3128 Nov 22 20:47 CODE_OF_CONDUCT.md
-rwxrwxrwx  1 vlad vlad   17863 Nov  6 18:10 configure.py
-rwxrwxrwx  1 vlad vlad    2129 Dec 15 12:43 config.yaml
-rwxrwxrwx  1 vlad vlad    8862 Nov 22 20:47 CONTRIBUTING.md
drwxrwxrwx  3 vlad vlad    4096 Nov  6 18:10 external
-rwxrwxrwx  1 vlad vlad 2632263 Dec 15 12:30 get-pip.py
drwxrwxrwx  8 vlad vlad    4096 Dec 15 12:23 .git
drwxrwxrwx  4 vlad vlad    4096 Nov 22 20:47 .github
-rwxrwxrwx  1 vlad vlad    1962 Nov 22 20:47 .gitignore
drwxrwxrwx  5 vlad vlad    4096 Nov 22 20:47 library
-rwxrwxrwx  1 vlad vlad   35149 Nov 22 20:47 LICENSE
-rwxrwxrwx  1 vlad vlad    2034 Dec 15 13:18 log.log
-rwxrwxrwx  1 vlad vlad    9407 Nov  6 18:10 main.py
-rwxrwxrwx  1 vlad vlad   12712 Nov 22 20:47 README.md
-rwxrwxrwx  1 vlad vlad    1439 Nov 22 20:47 requirements.txt
drwxrwxrwx  7 vlad vlad    4096 Nov  6 18:10 res
-rwxrwxrwx  1 vlad vlad     394 Nov 22 20:47 SECURITY.md
-rwxrwxrwx  1 vlad vlad    8164 Nov  6 18:10 simple-program.py
-rwxrwxrwx  1 vlad vlad   10081 Nov  6 18:10 theme-editor.py
drwxrwxrwx  3 vlad vlad    4096 Dec 15 12:48 tools
```

mathoudebine, 2024-06-18:
Documentation has been updated on the wiki.
Closing the issue for inactivity, tag me @mathoudebine in a comment if you want to reopen it

closed 2024-06-18 by mathoudebine