mathoudebine/turing-smart-screen-python #416: [ERROR] Python dependencies not installed on Fedora

AwekM, 2023-12-16:
**Describe the bug**  
I have followed the guide on how to start the program but it shows the error.

**Environment:**  
 - Smart screen model - Turing 3.5"
 - Revision of this project - 3.3.4
 - Fedora 39 (Linux)
 - Python version - Python 3.12
 - Hardware - Ryzen 5, RX5700XT

**Additional context**  

```
acapcap@acapcap:~/turing-smart-screen-python-3.3.4$ python --version
Python 3.12.0

acapcap@acapcap:~/turing-smart-screen-python-3.3.4$ python3 -m pip install -r requirements.txt
Defaulting to user installation because normal site-packages is not writeable
Ignoring numpy: markers 'python_version < "3.9"' don't match your environment
Ignoring GPUtil: markers 'python_version < "3.12"' don't match your environment
Collecting GPUtil@ git+https://github.com/mathoudebine/gputil.git@1.4.0-py3.12 (from -r requirements.txt (line 18))
  Cloning https://github.com/mathoudebine/gputil.git (to revision 1.4.0-py3.12) to /tmp/pip-install-7narjmcy/gputil_3c97b0d64d614c269fcf1da72e1ea025
  Running command git clone --filter=blob:none --quiet https://github.com/mathoudebine/gputil.git /tmp/pip-install-7narjmcy/gputil_3c97b0d64d614c269fcf1da72e1ea025
  Resolved https://github.com/mathoudebine/gputil.git to commit 1c6e890a5d9592d69ffcf50d31f12e89699b8c8e
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Ignoring pyadl: markers 'sys_platform == "win32"' don't match your environment
Ignoring pythonnet: markers 'sys_platform == "win32"' don't match your environment
Ignoring pywin32: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: Pillow~=10.1.0 in /usr/lib64/python3.12/site-packages (from -r requirements.txt (line 2)) (10.1.0)
Requirement already satisfied: pyserial~=3.5 in /usr/lib/python3.12/site-packages (from -r requirements.txt (line 3)) (3.5)
Requirement already satisfied: PyYAML~=6.0.1 in /usr/lib64/python3.12/site-packages (from -r requirements.txt (line 4)) (6.0.1)
Requirement already satisfied: psutil~=5.9.6 in /home/acapcap/.local/lib/python3.12/site-packages (from -r requirements.txt (line 5)) (5.9.6)
Requirement already satisfied: pystray~=0.19.5 in /home/acapcap/.local/lib/python3.12/site-packages (from -r requirements.txt (line 6)) (0.19.5)
Requirement already satisfied: babel~=2.13.1 in /home/acapcap/.local/lib/python3.12/site-packages (from -r requirements.txt (line 7)) (2.13.1)
Requirement already satisfied: ruamel.yaml~=0.18.5 in /home/acapcap/.local/lib/python3.12/site-packages (from -r requirements.txt (line 8)) (0.18.5)
Requirement already satisfied: sv-ttk~=2.5.5 in /home/acapcap/.local/lib/python3.12/site-packages (from -r requirements.txt (line 9)) (2.5.5)
Requirement already satisfied: numpy~=1.26.0 in /home/acapcap/.local/lib/python3.12/site-packages (from -r requirements.txt (line 13)) (1.26.2)
Requirement already satisfied: pyamdgpuinfo~=2.1.6 in /home/acapcap/.local/lib/python3.12/site-packages (from -r requirements.txt (line 21)) (2.1.6)
Requirement already satisfied: six in /usr/lib/python3.12/site-packages (from pystray~=0.19.5->-r requirements.txt (line 6)) (1.16.0)
Requirement already satisfied: python-xlib>=0.17 in /home/acapcap/.local/lib/python3.12/site-packages (from pystray~=0.19.5->-r requirements.txt (line 6)) (0.33)
Requirement already satisfied: setuptools in /usr/lib/python3.12/site-packages (from babel~=2.13.1->-r requirements.txt (line 7)) (67.7.2)
Requirement already satisfied: ruamel.yaml.clib>=0.2.7 in /home/acapcap/.local/lib/python3.12/site-packages (from ruamel.yaml~=0.18.5->-r requirements.txt (line 8)) (0.2.8)

acapcap@acapcap:~/turing-smart-screen-python-3.3.4$ sudo dnf install python3-tkinter
[sudo] password for acapcap: 
Updating and loading repositories:
Repositories loaded.
Package "python3-tkinter-3.12.0-1.fc39.x86_64" is already installed.

Nothing to do.

acapcap@acapcap:~/turing-smart-screen-python-3.3.4$ python3 configure.py
[ERROR] Python dependencies not installed. Please follow start guide: `https://github.com/mathoudebine/turing-smart-screen-python/wiki/System-monitor-:-how-to-start
```

mathoudebine, 2023-12-22:
@AwekM Can you edit the following lines in `configure.py`:
https://github.com/mathoudebine/turing-smart-screen-python/blob/73a7dd304e5e7f6b06a1746216c7893de3458f62/configure.py#L47-L59

Add `raise` just after the except l.53 (be careful of the 4 blank space before the word):
```python
except:
    raise
```

Then run back `python3 configure.py` and post the output here please?

AwekM, 2023-12-22:
After adding the `raise`, this is my output;

```
acapcap@acapcap:~/turing-smart-screen-python-3.3.4$ python3 configure.py
Traceback (most recent call last):
  File "/home/acapcap/turing-smart-screen-python-3.3.4/configure.py", line 51, in <module>
    from PIL import Image, ImageTk
ImportError: cannot import name 'ImageTk' from 'PIL' (/usr/lib64/python3.12/site-packages/PIL/__init__.py). Did you mean: 'Image'?
```

mathoudebine, 2023-12-22:
I see now! It seems you're missing some system packages, can you try `sudo dnf install python3-pillow-tk`?

AwekM, 2023-12-22:
Niceeee. That seems to be the solution!
Thanks for the help again!

closed 2023-12-22 by AwekM