mathoudebine/turing-smart-screen-python #419: Static data/image update on PC wake up

Sanichka, 2023-12-17:
**Is your feature request related to a problem? If so, please describe the problem.**  
After I wake up PC screen refreshing only parts with dynamic data(e.g sensors, percents etc) while background image is corrupted.

**Describe the feature / solution to your problem you'd like**  
Add backround and static elements refresh on PC woke up.

**Describe alternatives you've considered / and or tested**  
I haven't dived into code, so unfortunetly my only alternative is to manually restart program each day when I woke up PC.

**Screenshots / photos & mockups of the Turing screen**  
[image]
[image]

**Environment:**  
- Smart screen model UsbPCMonitor 5
- Revision of this project release 3.3.4
- OS with version Windows 10
- Python version Python 3.9.13
- Hardware AMD Ryzen 9 5900X, Nvidia 1070Ti

**Additional context**  
Logs from log.log file:
16.12.2023 15:52:34 [DEBUG] Using Python 3.9.13 (tags/v3.9.13:6de2ca5, May 17 2022, 16:36:42) [MSC v.1929 64 bit (AMD64)]
16.12.2023 15:52:34 [INFO] Tray icon has been displayed
16.12.2023 15:52:34 [INFO] Display reset (COM port may change)...
16.12.2023 15:52:39 [DEBUG] Static COM port: COM4
16.12.2023 15:52:39 [DEBUG] HW sub-revision: SubRevision.USBMONITOR_5
16.12.2023 15:52:39 [DEBUG] Drawing Image: BACKGROUND
16.12.2023 15:52:39 [DEBUG] Bitmap res/themes/5inchTheme2/background.png is now loaded in the cache
16.12.2023 15:52:39 [DEBUG] Drawing Text: DISK_USED_LABEL
16.12.2023 15:52:39 [DEBUG] Drawing Text: DISK_FREE_LABEL
16.12.2023 15:52:39 [DEBUG] Drawing Text: DISK_TOTAL_LABEL
16.12.2023 15:52:40 [DEBUG] Found one supported GPU: NVIDIA GeForce GTX 1070 Ti
16.12.2023 22:07:19 [DEBUG] Caught Windows window message event 536
16.12.2023 22:07:19 [INFO] Computer is going to sleep, display will turn off
17.12.2023 0:18:59 [DEBUG] Caught Windows window message event 536
17.12.2023 0:19:00 [DEBUG] Caught Windows window message event 536
17.12.2023 0:19:00 [INFO] Computer is resuming from sleep, display will turn on
17.12.2023 1:51:51 [DEBUG] Caught Windows window message event 536
17.12.2023 1:51:51 [INFO] Computer is going to sleep, display will turn off
17.12.2023 12:36:16 [DEBUG] Caught Windows window message event 536
17.12.2023 12:36:16 [DEBUG] Caught Windows window message event 536
17.12.2023 12:36:16 [INFO] Computer is resuming from sleep, display will turn on

Sanichka, 2023-12-21:
> add ` display.display_static_images()` ` display.display_static_text()` after ` display.turn_on()`
> 
> (after line 144 in main,py)

That helped, thanks!

mathoudebine, 2023-12-22:
Thanks for investigating, I can confirm this is a bug on the UsbPCMonitor 5inch display model only. When the display is turned off (using its own command) then turned on again, some part of the display will be black instead of the previous bitmap.
It seems related to an orientation issue (portrait/landscape)
I will implement the workaround from @GaryStar in the next version

closed 2023-12-22 by mathoudebine

GaryStar, 2023-12-22:
I have the same issue on a 3.5 display. But in my case, the USB power is turned off by the motherboard on Sleep.
I've added a new variable in config.yaml to control this (also updated configure.py). In case everything is alright on wake from sleep you don't want a full background refresh as it takes a little time and it's not updating the dynamic values while is doing it.