mathoudebine/turing-smart-screen-python #421: No data is displayed, only a static splash screen.

guglovich, 2023-12-18:
**Describe the bug**  
No data is displayed, only a static splash screen.

**To Reproduce**  
1.Granted the necessary rights so that there would be no error
    sudo chmod 666 /dev/ttyS0
2. Run with the specified /dev/ttyS0

**Expected behavior**  
No reaction. Not even an error.

**Screenshots / photos of the Turing screen**  
[image]

**Environment:**  
 - Smart screen model: I'm not sure exactly. A or B.
 - Revision of this project: main last
 - OS with version: Arch Linux
 - Python version: 3.11
 - Hardware: AMD Ryzen, Radeon RX460

mathoudebine, 2023-12-22:
Hi @guglovich 
Are you sure your smart screen is revision A?
You can check this wiki page to identify your model: https://github.com/mathoudebine/turing-smart-screen-python/wiki/Hardware-revisions

The revision A hardware usually have a serial port named `/dev/ttyACM[X]` on Linux platforms.
Can you list the COM ports you have in '/dev'?
Can you try the COM port auto discovery? `COM_PORT: AUTO` in config.yaml

guglovich, 2023-12-23:
@mathoudebine 

Judging by the picture, I have revision B. But when selecting B, the app didn't even attempt to connect or search. When selecting A and /dev/ttyS0, the application makes several attempts.

Automatic port selection was not working for me.

`lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS0/device/driver -> ../../../bus/pnp/drivers/serial
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS10/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS11/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS12/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS13/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS14/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS15/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS16/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS17/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS18/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS19/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS1/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS20/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS21/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS22/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS23/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS24/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS25/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS26/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS27/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS28/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS29/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS2/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS30/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS31/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS3/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS4/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS5/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS6/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS7/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS8/device/driver -> ../../../bus/platform/drivers/serial8250
lrwxrwxrwx 1 root root 0 дек 23 00:25 /sys/class/tty/ttyS9/device/driver -> ../../../bus/platform/drivers/serial8250`

mathoudebine, 2024-01-10:
The `/dev/ttyS[...]` are serial ports from your system, they are not the serial port of the screen
First you need to find the correct serial port of your display:
Unplug your display, run `ls /dev/tty*` then plug your display and run the same command. Compare the 2 output to find the serial port that has appeared in the second output: this is your display serial port

guglovich, 2024-01-12:
@mathoudebine ls /dev/tty* with screen enabled and disabled, gives two identical outputs

mathoudebine, 2024-01-12:
Can you try again but using another USB cable? And maybe also try another USB port of your computer

guglovich, 2024-01-12:
> Can you try again but using another USB cable? And maybe also try another USB port of your computer

The device itself is visible in the lsusb
Bus 005 Device 003: ID 0483:0065 STMicroelectronics llTechCo.,Ltd

mathoudebine, 2024-01-28:
Are you sure this device is your screen? Does it disappear when you unplug the screen from the computer?

If this device is your screen, then I'm not sure it is supported by this project. There is a list of existing displays and their output from `lsusb` here: https://github.com/mathoudebine/turing-smart-screen-python/tree/main/tools/lsusb
None of the supported screen is identified as `STMicroelectronics llTechCo.,Ltd`
Where did you buy it?

guglovich, 2024-01-28:
> Are you sure this device is your screen? Does it disappear when you unplug the screen from the computer?
>
> If this device is your screen, then I'm not sure it is supported by this project. There is a [...]

Yes, I found that out by plugging and unplugging the device.

Bought from a Chinese, a certain Fujian Lesidi Electronic Commerce Co., Ltd.

mathoudebine, 2024-01-28:
If you have a link of the website where you bought it, I am interested.
It can be a new smart screen model that we did not integrate yet to this project

guglovich, 2024-01-28:
> If you have a link of the website where you bought it, I am interested. It can be a new smart screen model that we did not integrate yet to this project

[https://www.ozon.ru/product/monitor-pk-3-5-dyuymovyy-ips-usb-mini-ekran-cpu-ram-hdd-displey-temperatury-dannyh-360-gradusnyy-902618658/](url)

The site is not multinational. I tried to find this seller on AliExpress by name, but without success. Other sellers on AliExpress are also not found or they have empty stores with no products.

mathoudebine, 2024-01-29:
From the product pictures it looks like a Revision A screen from Turing brand, but they may have reused pictures from another website...
Is the screen provided with a Windows software? Can you share a link to download it, or a zip file? You can drag-and-drop files in a comment to attach it.

guglovich, 2024-01-29:
> From the product pictures it looks like a Revision A screen from Turing brand, but they may have reused pictures from another website... Is the screen provided with a Windows software? Can you share a link to download it, or a zip file? You can drag-and-drop files in a comment to attach it.

https://www.turzx.com/
The Chinese utility from this site is used for this product.

mathoudebine, 2024-02-25:
This is very strange, maybe your display is broken? If it is turzx.com software then the display is a Turing Smart Screen, but it should appear as /dev/ttyACM0 and should be listed from lsusb as:
```
Bus 001 Device 002: ID 1a86:5722 QinHeng Electronics UsbMonitor
```
Do you have a Windows computer to try to use the turzx software with your display?

guglovich, 2024-03-23:
I finally figured it out today after reading customer comments. I downloaded the SmartMonitor.exe software from another person and it started showing some data. So I got the screen you have labeled as (Fuldho?) 3.5" IPS Screen

I take it that your program cannot be implemented for that model?

mathoudebine, 2024-04-29:
Thanks for the heads up, indeed the Fuldho screen cannot work with this program because it has a different behavior than Turing/Xuanfang/other smart screens.
The Fuldho screen receives sensor data (numeric values) then display the content (text, charts) on its own, whereas for Turing/Xuanfang the Smart Screen program does the content drawing and sends bitmaps to the screen.

closed 2024-04-29 by mathoudebine