mathoudebine/turing-smart-screen-python #426: New 3.5" screen variant?

braewoods, 2023-12-21:
I found this while looking around for different models of USB 3.5" screens. I found this:

https://www.aliexpress.us/item/3256806022857282.html

It mentions some software called Guition which seems to be different from the stuff I have already seen. Maybe this is another variant of a newer screen type?

mathoudebine, 2023-12-22:
Hello @braewoods you're right it seems there is a new smart screen model on the market, thanks for telling me!
Damn, this will not help with my smart screen addiction 😆
I will definitely buy some and try to integrate it to this project!

[image]
This line in the ad is a lie, or this smart screen use a different technology than the others 3.5" that are unable to display gifs due to their very low refresh rate.

braewoods, 2023-12-25:
> This line in the ad is a lie, or this smart screen use a different technology than the others 3.5" that are unable to display gifs due to their very low refresh rate.

It could be true if the raw USB is using USB 2.0 instead of 1.1. VoCore screens can handle real time updates due to using USB 2.0. Or it may be a model that uploads everything to the internal storage and operates similar to the HID screens I am still working on.

braewoods, 2024-01-22:
I received the screen finally. Out of the box I find that it has a USB type C port and a rectangular opening I assume is for an SD card. I will examine it more thoroughly later to find how it works.

braewoods, 2024-01-22:
I got a dump of the `lsusb` stuff. The endpoints make me scratch my head right now.

```Bus 001 Device 008: ID 303a:1001 Espressif USB JTAG/serial debug unit
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x303a 
  idProduct          0x1001 
  bcdDevice            1.01
  iManufacturer           1 Espressif
  iProduct                2 USB JTAG/serial debug unit
  iSerial                 3 EC:DA:3B:CE:19:24
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0062
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       0 
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      0 
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      CDC Call Management:
        bmCapabilities       0x03
          call management
          use DataInterface
        bDataInterface          1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      2 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol      1 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0001
  Self Powered```

mathoudebine, 2024-02-19:
Hi again! I finally received the product and investigated a bit using HHD Serial Port Monitor to get the data transiting between the display and the computer.
I have bad news unfortunately.

Based on what I've seen, it seems like the display is based on an ESP32. When a theme is selected from the Windows software, a new complete ESP32 firmware is compiled by the Windows software and flashed to the display. I can see it on the Serial Port Monitor interface.

Then, when program is running, only sensors data are sent to the display periodically as you can see on this screen capture:
[image]
Frames in red are sent periodically by the software to the display, and contains a set of predefined sensor values separed by `$N`. Here I created a theme with only 1 sensor (CPU usage) and it is the data you can see (22,61) in the frame. The display acknowledge every frame from the computer.

So it seems you were right: even though the display does not use the HID protocol to communicate with the computer but a virtual serial port, it is working as the Fuldho 3.5" IPS Screen you're working on.
I don't think it can be integrated to this project.

braewoods, 2024-02-19:
I appreciate the information you were able to provide. I am currently working on a refactoring of my own solution to the Fuldho screens that may be amenable to this screen in the future but for now isn't a priority. Right now I am expanding the original software I wrote to work with the newer screen variants by the same vendor which use a new serial protocol instead of HID. There's apparently a 2.1" screen but so far I have only seen the 5" screen floating around Aliexpress.

closed 2024-03-01 by mathoudebine

matu6968, 2024-06-29:
> Hi again! I finally received the product and investigated a bit using HHD Serial Port Monitor to get the data transiting between the display and the computer. I have bad news unfortunately.
>
> Based [...]

It seems like everything including the toolchain, obfuscated (maybe?) source code is in the main executable by finding patterns like GCC using a hex editor, also will try using binwalk to see if there is anything inetresting in there. So to add support, reverse engineering of the executable is needed to extract the source code and the build flags.

q20021410, 2026-09-14:
Hi @mathoudebine and everyone interested in this screen! 👋

Like many others, I purchased this exact **Guition 3.5" IPS USB-C Secondary Screen**, but the proprietary software bundled by the seller was no longer functioning properly. To bring this hardware back to life and make it genuinely useful, I refactored the ESP32-C3 firmware and transmission logic with the assistance of AI, and I can confirm it now **works flawlessly with `turing-smart-screen-python`**!

I created a dedicated repository containing the pre-compiled firmware binaries, factory restore backups, and full setup guides:
👉 **[Guition-Turing-Smart-Screen (GitHub)](https://github.com/q20021410/Guition-Turing-Smart-Screen)**

---

### 📷 Hardware Showcase (Verified Working)

| 🖥️ Real-Time Monitoring in Action (Landscape) | ⏳ 80MHz Custom Firmware Standby Screen |
| :---: | :---: |
| [image] | [image] |

---

### 🔍 Technical Details & How It Works

- **Hardware**: ESP32-C3 RISC-V @ 160MHz, 4MB Flash, ST7796S 3.5" IPS display ([AliExpress Link](https://aliexpress.com/item/1005006169962183.html)).
- **Protocol Compatibility**: By flashing the 80MHz SPI DMA custom firmware at offset **`0x10000`**, the screen responds 1:1 to the standard **`REVISION: A`** (Turing Rev A / Command 197 `DISPLAY_BITMAP`) protocol.
- **Factory Firmware Backup**: A clean 4MB raw factory flash dump is also provided at `0x0`, so anyone can safely restore it at any time.

### ⚙️ Recommended `config.yaml` Settings

```yaml
display:
  REVISION: A
  DISPLAY_REVERSE: true
  RESET_ON_STARTUP: false

matu6968, 2026-09-14:
> Hi @mathoudebine and everyone interested in this screen! 👋
>
> Like many others, I purchased this exact **Guition 3.5" IPS USB-C Secondary Screen**, but the proprietary software bundled by the seller [...]

No source code for both ends and with it being apparently closed source (only .exe binaries and pre-compiled firmware binaries for the ESP32-C3 display) based off upstream sounds like a GPL violation to me, will sources be published (atleast for the upstream frontend software fork) to comply with the GPL?

q20021410, 2026-09-14:
> > Hi [@mathoudebine](https://github.com/mathoudebine) and everyone interested in this screen! 👋
> > Like many others, I purchased this exact **Guition 3.5" IPS USB-C Secondary Screen**, but the proprietary software bundled by the seller was no longer functioning properly. To bring this hardware back to life and make it genuinely useful, I refactored the ESP32-C3 firmware and transmission logic with the assistance of AI, and I can confirm it now **works flawlessly with `turing-smart-screen-python`**!
> > I created a dedicated repository containing the pre-compiled firmware binaries, factory restore backups, and full setup guides:
> > 👉 **[Guition-Turing-Smart-Screen (GitHub)](https://github.com/q20021410/Guition-Turing-Smart-Screen)**
> > 
> > ### 📷 Hardware Showcase (Verified Working)
> > 
> > 
> > 
> > 🖥️ Real-Time Monitoring in Action (Landscape)
> > ⏳ 80MHz Custom Firmware Standby Screen
> > 
> > 
> > 
> > 
> > [image]
> > [image]
> > 
> > 
> > 
> > 
> > ### 🔍 Technical Details & How It Works
> > 
> > * **Hardware**: ESP32-C3 RISC-V @ 160MHz, 4MB Flash, ST7796S 3.5" IPS display ([AliExpress Link](https://aliexpress.com/item/1005006169962183.html)).
> > * **Protocol Compatibility**: By flashing the 80MHz SPI DMA custom firmware at offset **`0x10000`**, the screen responds 1:1 to the standard **`REVISION: A`** (Turing Rev A / Command 197 `DISPLAY_BITMAP`) protocol.
> > * **Factory Firmware Backup**: A clean 4MB raw factory flash dump is also provided at `0x0`, so anyone can safely restore it at any time.
> > 
> > ### ⚙️ Recommended `config.yaml` Settings
> > display:
> >   REVISION: A
> >   DISPLAY_REVERSE: true
> >   RESET_ON_STARTUP: false
> 
> No source code for both ends and with it being apparently closed source (only .exe binaries and pre-compiled firmware binaries for the ESP32-C3 display) based off upstream sounds like a GPL violation to me, will sources be published (atleast for the upstream frontend software fork) to comply with the GPL?

Yes, I plan to upload it separately.

q20021410, 2026-09-14:
> No source code for both ends and with it being apparently closed source (only .exe binaries and pre-compiled firmware binaries for the ESP32-C3 display) based off upstream sounds like a GPL [...]

The change is complete. I also separated and re-uploaded the entire source code and the ESP-related parts.