mathoudebine/turing-smart-screen-python #431: Add weather info

XLave, 2023-12-31:
Is there a way to display the local weather info like in UsbMonitor.exe? Would be a great addition.

**Is your feature request related to a problem? If so, please describe the problem.**  
A description of what the problem is. E.g. I cannot display this data the way I want [...]

**Describe the feature / solution to your problem you'd like**  
A description of what you want to happen.

**Describe alternatives you've considered / and or tested**  
A description of any alternative solutions or features you've considered.

**Screenshots / photos & mockups of the Turing screen**  
Add screenshots or photos/mockups of the rendering on the Turing screen to help explain your request.
You can drag and drop photos here to add them to the description.

**Environment:**  
 - Revision of this project [e.g. 1.1.1, `main` branch, specific commit]
 - OS with version [e.g. Windows 11, Ubuntu 22.04]
 - Python version [e.g. Python 3.8]
 - Hardware [e.g. Intel CPU, Nvidia GPU, Raspberry Pi 3 B+ ...]

**Additional context**  
Add any other context or screenshots about the feature request here.

BlackHargaan, 2024-01-01:
There is a discussion post where someone added it. It hasn't been merged yet, and there were some problems when I just copied the code, but I used the guy's concepts to add it to mine, and it works.

XLave, 2024-01-01:
Thx, I didn't know it. I't trying to use it but I get the error:
\library\scheduler.py", line 157, in <module>
    @schedule(timedelta(seconds=config.THEME_DATA['STATS']['WEATHER'].get("INTERVAL", None)).total_seconds())
                                ~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^
KeyError: 'WEATHER'
How can I fix it?

GaryStar, 2024-01-01:
> Thx, I didn't know it. I't trying to use it but I get the error: \library\scheduler.py", line 157, in @schedule(timedelta(seconds=config.THEME_DATA['STATS']['WEATHER'].get("INTERVAL", None)).total_seconds()) ~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^ KeyError: 'WEATHER' How can I fix it?

You need to add a WEATHER section in your theme (in STATS section):
something like this:
```
  WEATHER:
    INTERVAL: 600
    TEXT:
      SHOW: True
      X: 35
      Y: 52
      FONT: roboto-mono/RobotoMono-Bold.ttf
      FONT_SIZE: 18
      FONT_COLOR: 150, 200, 200
      BACKGROUND_IMAGE: background.png
```

XLave, 2024-01-01:
Yeah, thx, I've just tried it, now it shows the info in the display.
I have another issue:
 File "\turing-smart-screen-python-feature-names-alert-weather\library\stats.py", line 33, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'.
If i delete that line it boot up, but later I enounter another issue related and can't start it.

GaryStar, 2024-01-01:
You need to install ```requests``` module by yourself (it's now added in ```requirements.txt``` so on a fresh install it will be automatically added)
This module is used to fetch data from OpenWeatherMap. It's a standard module to fetch data from external sites.

To install it you can use:
```pip install requests```
or 
```python -m pip install requests```
or 
```py.exe -m pip install requests```

XLave, 2024-01-01:
Thx for your quick support!

I have installed that module and it fixes but I have another issue with LibreHardware.dll related:
File "...\turing-smart-screen-python-feature-names-alert-weather\library\sensors\sensors_librehardwaremonitor.py", line 42, in <module>
    clr.AddReference(lhm_dll)
System.IO.FileLoadException: No se puede cargar el archivo o ensamblado 'file:///...\turing-smart-screen-python-feature-names-alert-weather\external\LibreHardwareMonitor\LibreHardwareMonitorLib.dll' ni una de sus dependencias. Operación no admitida. (Excepción de HRESULT: 0x80131515)
Nombre de archivo: 'file:///...\turing-smart-screen-python-feature-names-alert-weather\external\LibreHardwareMonitor\LibreHardwareMonitorLib.dll' ---> System.NotSupportedException: Se intentó cargar un ensamblado desde una ubicación de red, por lo que el ensamblado habría sido incluido en un espacio aislado de versiones anteriores de .NET Framework. Esta versión de .NET Framework no habilita la directiva CAS de forma predeterminada, por lo que esta carga puede ser peligrosa. Si esta carga no va a incluir el ensamblado en un espacio aislado, habilite el modificador loadFromRemoteSources. Vea http://go.microsoft.com/fwlink/?LinkId=155569 para obtener más información.
   en System.Reflection.RuntimeAssembly._nLoad(AssemblyName fileName, String codeBase, Evidence assemblySecurity, RuntimeAssembly locationHint, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   en System.Reflection.RuntimeAssembly.InternalLoadAssemblyName(AssemblyName assemblyRef, Evidence assemblySecurity, RuntimeAssembly reqAssembly, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   en System.Reflection.RuntimeAssembly.InternalLoadFrom(String assemblyFile, Evidence securityEvidence, Byte[] hashValue, AssemblyHashAlgorithm hashAlgorithm, Boolean forIntrospection, Boolean suppressSecurityChecks, StackCrawlMark& stackMark)
   en System.Reflection.Assembly.LoadFrom(String assemblyFile)
   en Python.Runtime.AssemblyManager.LoadAssemblyFullPath(String name)
   en Python.Runtime.CLRModule.AddReference(String name)

If I use the main branch the monitor works great, but I want to use your branch because of the redrawn from waking up and the weather info. Sorry for the trouble, I don't know Python nor any programming language.

GaryStar, 2024-01-01:
if you use LibreHardware you need to start main.py with admin rights.

XLave, 2024-01-01:
I was running it with admin rights, but the problem was another: https://www.jesusninoc.com/06/16/posible-solucion-al-error-se-intento-cargar-un-ensamblado-desde-una-ubicacion-de-red-por-lo-que-el-ensamblado-habria-sido-incluido-en-un-espacio-aislado-de-versiones-anteriores-de-net-framework/

Now I fixed it and it's running great!! Thx for your time, happy 2024!!

XLave, 2024-01-01:
I'm trying to add a weather icon from OpenWeather API by adding the line "icon = data['weather'][0]['icon']" to the stats.py but i'm only able to get the icon code instead of the icon (https://openweathermap.org/weather-conditions). Is it possible to fetch an image from a url to display it on the monitor? How can I achieve that?

GaryStar, 2024-01-01:
those icons are static and not so many. maybe they can all be manually downloaded in the theme folder (for example) and used from there based of the code received. I will try to implement something like this in the next days..

GaryStar, 2024-01-04:
I've updated the code with support for weather icons as well
There are two sets of icons added:
 - one as the default set that will be used on any theme 
 - one in theme folder that will be used instead of default one (separation is done for each icon, so you can have only a couple of custom icons and the rest will be used from the default folder)
 
Example usage:
```
  WEATHER:
    INTERVAL: 600
    TEXT:
      SHOW: True
      X: 30
      Y: 52
      FONT: roboto-mono/RobotoMono-Bold.ttf
      FONT_SIZE: 18
      FONT_COLOR: 150, 200, 200
      BACKGROUND_IMAGE: background.png
    ICON:
      SHOW: True
      X: 30
      Y: 11
      SCALE: 0.43
      BACKGROUND_IMAGE: background.png
```
[image]
[image]
[image]

XLave, 2024-01-04:
Thanx so much! It works perfect now with weather icons, that was a fast implementation.

XLave, 2024-01-10:
Hi, I'm having an "connection error" somentimes when geeting back from sleep in the weather info, any thoughs about how to fix it?

GaryStar, 2024-01-10:
yes.. add
```
time.sleep(5)  # on wake, network access is not ready immediately
```
before
```
stats.Weather.stats()  # refresh is needed because time between automated refreshes is usually over 10-15 minutes
```
in ```main.py```

P.S. I've updated my branch with this change. I also added code for the smart screen to be turned on and off (not off completely just brightness set to 0) when system is turning the main monitor on/off (not shutdown or sleep)
My branch is here: https://github.com/GaryStar/turing-smart-screen-python/tree/feature/names-alert-weather

ds0tr0, 2025-05-01:
uau! I would like to add this to my theme... but when i try on simulated screen. the icon doesnt appear
```
WEATHER:
    # For optimal use, if you want to stay on the free plan (1000 calls), the interval should be 90 MINIMUM (not really useful as the API didn't update that quickly)
    INTERVAL: 600
    ICON:
       SHOW: True
       X: 30
       Y: 11
       SCALE: 0.43
       BACKGROUND_IMAGE: background.png
    TEMPERATURE:
      TEXT:
        SHOW: True
        X: 205
        Y: 24
        WIDTH: 60   # Uncomment to force a static width
        HEIGHT: 14   # Uncomment to force static height
        FONT: generale-mono/GeneraleMonoA.ttf
        FONT_SIZE: 14
        FONT_COLOR: 255, 255, 255
        #BACKGROUND_COLOR: 132, 154, 165
        BACKGROUND_IMAGE: background.png
        ALIGN: left  # left / center / right
        ANCHOR: lt   # Check https://pillow.readthedocs.io/en/stable/handbook/text-anchors.html
    WEATHER_DESCRIPTION:
      TEXT:
        SHOW: True
        X: 205
        Y: 10
        WIDTH: 75   # Uncomment to force a static width
        HEIGHT: 14   # Uncomment to force static height
        FONT: generale-mono/GeneraleMonoA.ttf
        FONT_SIZE: 14
        FONT_COLOR: 2, 216, 243
        #BACKGROUND_COLOR: 132, 154, 165
        BACKGROUND_IMAGE: background.png 
        ALIGN: left  # left / center / right
        ANCHOR: lt   # Check https://pillow.readthedocs.io/en/stable/handbook/text-anchors.html 
```