mathoudebine/turing-smart-screen-python #439: Adding long string of text fails

n-i-x, 2024-01-06:
**Describe the bug**  
Adding text to an image fails with the error:

```
Traceback (most recent call last):
  File "/workspaces/turing-smart-screen-python/button_card.py", line 50, in <module>
    lcd_comm.DisplayText(game['gamename'], 5, 5,
  File "/workspaces/turing-smart-screen-python/library/lcd/lcd_comm.py", line 273, in DisplayText
    self.DisplayPILImage(text_image, left, top)
  File "/workspaces/turing-smart-screen-python/library/lcd/lcd_comm_rev_c.py", line 303, in DisplayPILImage
    img, pyd = self._generate_update_image(image, x, y, Count.Start, Command.UPDATE_BITMAP,
  File "/workspaces/turing-smart-screen-python/library/lcd/lcd_comm_rev_c.py", line 365, in _generate_update_image
    payload.extend(bytearray.fromhex(image_size))
ValueError: non-hexadecimal number found in fromhex() arg at position 5
```

**To Reproduce**  
Steps to reproduce the behavior:
```
lcd_comm.DisplayText("1000 Miglia: Great 1000 Mile", 5, 5,
                                font="geforce/GeForce-Bold.ttf",
                                font_size=60,
                                font_color=(255, 255, 255),
                                background_image=background,
                                align="left",
                                anchor="lt")
```

`image_size` from line 365 in _generate_update_image = '10ff3'. It appears that when the value is more than 4 bytes there is an error.

**Expected behavior**  
Text should render on the screen.

**Screenshots / photos of the Turing screen**  
Nothing displays.

**Environment:**  
 - Smart screen model Turing 5
 - Revision of this project `main` branch
 - OS with version OSX 14.2.1
 - Python version 3.9
 - Hardware MacBook Pro M3 Pro

**Additional context**  
I attempted to zero pad the string but that didn't work. Reducing the string below the size that ends up being 5 bytes works fine.

gwendal-h, 2024-01-07:
Yes i fixed this one here 46f5398aee28081cd4789ae2a834ac9ba55f134f

mathoudebine, 2024-01-28:
Hello, it has been fixed on the `main` branch.
Thanks @gwendal-h I've backported your commit until your PR is merged

closed 2024-01-28 by mathoudebine