mathoudebine/turing-smart-screen-python #453: AttributeError: 'NoneType' object has no attribute 'get'

luminesupreme, 2024-01-16:
*C:\Users\lumi\Downloads\turing-smart-screen-python-3.3.4\turing-smart-screen-python-3.3.4>python.exe .\configure.py
Traceback (most recent call last):
  File "C:\Users\lumi\Downloads\turing-smart-screen-python-3.3.4\turing-smart-screen-python-3.3.4\configure.py", line 434, in <module>
    configurator = TuringConfigWindow()
                   ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\lumi\Downloads\turing-smart-screen-python-3.3.4\turing-smart-screen-python-3.3.4\configure.py", line 240, in __init__
    self.load_config_values()
  File "C:\Users\lumi\Downloads\turing-smart-screen-python-3.3.4\turing-smart-screen-python-3.3.4\configure.py", line 278, in load_config_values
    self.load_theme_preview()
  File "C:\Users\lumi\Downloads\turing-smart-screen-python-3.3.4\turing-smart-screen-python-3.3.4\configure.py", line 259, in load_theme_preview
    author_name = theme_data.get('author', 'unknown')
                  ^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'

[image]

**Environment:**  
 - Smart screen model [Turing 3.5”]
 - Revision of this project [3.3.4]
 - OS with version [Windows 11 Pro]
 - Python version [Python 3.12]
 - Hardware [AMD Ryzen 7 7700X]

**Additional context**  
Hello, sorry if this is a stupid question but I'm pretty new to tech and improving my setup here. This is the error I'm getting when I attempt step 4- starting the configuration wizard. I was able to get this far following a YouTube tutorial, but this error isn't mentioned.

mathoudebine, 2024-01-28:
Hello @luminesupreme
This is a very starnge error that should not happen.
Can you post the content of the `config.yaml` file here?
Did you manually edit some Python files or theme files by yourself?

Ergonautaa, 2024-04-03:
I have this same problem, additionally I can share that yesterday i actually opened the configuration wizard and tryed many settings and averything worked perfectly fine, but when i tried to make the task for the screen to work everytime i logged on, the task didnt work and once i restarted the PC i get this exact attribute error

mathoudebine, 2024-04-26:
To fix this issue, you should replace the `config.yaml` file with the original one. The you can restart the `configure.py` properly without error

closed 2024-04-28 by mathoudebine