mathoudebine/turing-smart-screen-python #471: RX 5500 xt no memory suported

ibonFPV, 2024-02-17:
Hi 

Sorry, but i dont know if i doinfg something worng or my rx5500 xt is not suported, that is the error: thanks

17/02/2024 11:26:39 [INFO] Tray icon has been displayed
17/02/2024 11:26:39 [INFO] Display reset (COM port may change)...
17/02/2024 11:26:44 [DEBUG] Static COM port: COM3
17/02/2024 11:26:44 [DEBUG] HW sub-revision: SubRevision.USBMONITOR_3_5
17/02/2024 11:26:44 [DEBUG] Drawing Image: BACKGROUND
17/02/2024 11:26:44 [DEBUG] Bitmap res/themes/LandscapeMagicBlue/background.png is now loaded in the cache
17/02/2024 11:26:44 [DEBUG] Drawing Text: DISK_LABEL
17/02/2024 11:26:44 [DEBUG] Drawing Text: DISK_USED_LABEL
17/02/2024 11:26:45 [DEBUG] Found one supported GPU: ASUS Radeon RX 5500 XT
17/02/2024 11:26:45 [WARNING] Your GPU memory relative usage (%) is not supported yet

GPU - ASUS Radeon RX 5500 XT - Principal/Discreto
VRAM - 8176 MB - GDDR6 1750 MHz
Versión de controlador - 23.40.02-240111a-400139C-AMD-Software-Adrenalin-Edition
Versión del controlador de Windows de AMD - 31.0.24002.92

ibonFPV, 2024-02-18:
I dont know why it dont get the total memory of gpu but i put it directly in de code 

class Gpu:
    @staticmethod
    def stats():
        load, memory_percentage, memory_used_mb, temperature = sensors.Gpu.stats()
        fps = sensors.Gpu.fps()
        display_gpu_stats(load, (**(memory_used_mb / 8176**) * 100), memory_used_mb, temperature, fps)

mathoudebine, 2024-02-19:
Hi @ibonFPV you can ignore the warning message.
This is because AMD GPUs do not offer a way to get the total RAM amount automatically. This is a feature for Nvidia GPUs only.
If you add it manually in the code like you did, it will work fine
I'm closing this issue but feel free to comment if you need more help

closed 2024-02-19 by mathoudebine