mathoudebine/turing-smart-screen-python #473: GPU and fps fan error

DJRoby19, 2024-02-20:
20/02/2024 18:05:13 [DEBUG] HW sub-revision: SubRevision.TURING_3_5
20/02/2024 18:05:13 [DEBUG] Drawing Image: BACKGROUND
20/02/2024 18:05:13 [DEBUG] Bitmap res/themes/Cyberdeck/background.png is now loaded in the cache
20/02/2024 18:05:13 [INFO] Detected AMD GPU(s)
20/02/2024 18:05:13 [WARNING] Your CPU Fan Speed is not supported yet
20/02/2024 18:05:13 [WARNING] Your GPU FPS is not supported yet

**Environment:**  
 - Smart screen model [e.g. Turing 5", XuanFang 3.5”]
 - Revision of this project [e.g. 3.4.0, `main` branch, specific commit]
 - OS with version [Linux, Ubuntu 22.04]
 - Python version [e.g. Python 3.10]
 - Hardware [e.g. AMD CPU, Amd GPU,RX7800XT]

When I launch the Cyberdek theme it gives me this problem

mathoudebine, 2024-02-20:
Can you open a python shell from a terminal and type these commands?
```python
import psutil
print(psutil.sensors_fans())
```
What do you get?

DJRoby19, 2024-02-20:
roberto  /opt/turing-smart-screen-python-main  2  python3
Python 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import psutil
>>> print(psutil.sensors_fans())
{'amdgpu': [sfan(label='', current=0)]}
>>>

mathoudebine, 2024-02-20:
Thanks for testing it!
As you can see only your AMD GPU fan was detected, which means the psutil library did not detect the CPU fan sensor from your motherboard.
As for the FPS it is only available for Windows platform for now.
So you can ignore the 2 warnings, the program will work fine but you won't be able to display CPU fan speed or FPS on themes unfortunately

DJRoby19, 2024-02-21:
So I tried changing kernel from 5.15 to 6.5 now it gives me this

roberto  ~  python3
Python 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import psutil
>>> print(psutil.sensors_fans())
{'amdgpu': [sfan(label='', current=0)], 'nct6798': [sfan(label='', current=748), sfan(label='', current=885), sfan(label='', current=0), sfan(label='', current=734), sfan(label='', current=0), sfan(label='', current=1454), sfan(label='', current=864)]}
>>>

mathoudebine, 2024-02-25:
Interesting! I have quite the same sensors on my side:
```sh
>$ inxi -s
Sensors:
  System Temperatures: cpu: 39.5 C mobo: 29.0 C gpu: amdgpu temp: 48.0 C
  Fan Speeds (rpm): fan-1: 736 fan-2: 969 fan-3: 718 fan-4: 699 fan-5: 0
    fan-6: 0 fan-7: 0 gpu: amdgpu fan: 1708

>$ python
Python 3.11.7 (main, Jan 29 2024, 16:03:57) [GCC 13.2.1 20230801] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import psutil
>>> print(psutil.sensors_fans())
{'nct6798': [sfan(label='', current=740), sfan(label='', current=948), sfan(label='', current=707), sfan(label='', current=685), sfan(label='', current=0), sfan(label='', current=0), sfan(label='', current=0)], 'amdgpu': [sfan(label='', current=1708)]}
```
I had to run `sudo sensors-detect` first to get the complete list from psutil.

The nct6798 is the Nuvoton's Super I/O controller that manages HW monitoring on AMD Ryzen platforms.
The issue is: how can we know which fan is the CPU fan? Do you know on your side if the fan1 is the CPU fan?

mathoudebine, 2024-02-25:
And there is also another issue: for the fan speeds from nct6798 there is no max. speed info, so it is not possible to compute a percentage:
```sh
>$ sensors
nct6798-isa-0290
Adapter: ISA adapter
fan1:                      707 RPM  (min =    0 RPM)
fan2:                      906 RPM  (min =    0 RPM)
fan3:                      691 RPM  (min =    0 RPM)
fan4:                      657 RPM  (min =    0 RPM)
fan5:                        0 RPM  (min =    0 RPM)
fan6:                        0 RPM  (min =    0 RPM)
fan7:                        0 RPM  (min =    0 RPM)

[...]

amdgpu-pci-0800
Adapter: PCI adapter 
fan1:         825 RPM  (min =    0 RPM, max = 3200 RPM)
```

DJRoby19, 2024-02-25:
In my opinion and fan1 however read here and interesting possible solution
https://github.com/ksk0/cpu-fan

mathoudebine, 2024-02-26:
> To identify the controller, script will speed up any fan attached to controller, and your task will be to see (or hear) when CPU FAN speeds up. When you detect it, answer affirmative to the script.

This is a creative way to identify the CPU fan I agree! Unfortunately, it is not automated and I don't want to add a manual step to the program and change the fan speed. Here is what i think can be implemented:

- on Linux platforms, add a warning to run `sudo sensors-detect` to enable full sensors detection
- on the Configuration wizard, add a dropdown list to ask user to select which fan is the CPU fan. The list will have all fan displayed with their current speed, but it is up to the user to identify which one is the CPU fan.

DJRoby19, 2024-02-26:
It seems like a great idea anyway my cpu-fan and fan1

mathoudebine, 2024-03-10:
I made a PR #484 for the changes: now on Linux the Configuration UI has a new field "CPU fan" to allow the user to select it. The list has all the fans and their current speed in % and in RPM, to help user select.
[image]

There is also a tooltip in case user is missing fans from the list, to run the sensors-detect command
[image]

DJRoby19, 2024-03-10:
Great, great news, thank you
Will this change work in the next version?

closed 2024-06-18 by mathoudebine