mathoudebine/turing-smart-screen-python #475: AX206 based display support

nvx, 2024-02-22:
I've recently picked up a screen which works with AIDA64 using AX206 LCD support via libusb, but it doesn't look like it's supported with this python code yet.

It's available on AliExpress https://www.aliexpress.com/item/1005005600240555.html and while it looks similar to the screen mentioned in #426 I note that one mentions it works without AIDA64 so I imagine it's different (I couldn't see a link to download the software for that one to test if it worked on my AX206 based screen)

mathoudebine, 2024-02-22:
Hello, if your screen really is an AX206 then unfortunately it won't work with this project because we didn't implement the support yet.
But based on what I see from the reviews, it may also be a Turing 3.5 inch. It is not clear if the seller has several models, or if he doesn't really know what he is selling.
If you want, you can try the System Monitor program from this project and select "Turing 3.5" as screen model, and let me know if it works for you!

nvx, 2024-02-25:
> If you want, you can try the System Monitor program from this project and select "Turing 3.5" as screen model, and let me know if it works for you!

I did try all the 3.5" options and no dice. I suspect it is really an AX206 as selecting that in AIDA64 worked

mathoudebine, 2024-03-01:
I was thinking we could maybe add support for Linux first using the lcd4linux library, because people on Windows can at least use Aida64 which is far better than the Windows apps made for the serial smart screens

AquaFrostByte, 2024-11-24:
lcd4linux is broken at least the arch package and there is not realy a alternative that is easy to use :/