mathoudebine/turing-smart-screen-python #476: The parameters(load, temp, memory) of the Intel UHD 770 GPU are not displayed

ingenegra, 2024-02-23:
**Describe the bug**  
The parameters(load, temp, memory) of the Intel UHD 770 GPU are not displayed

**To Reproduce**  
Steps to reproduce the behavior:
1. Run main.py
2. See error in IDLE window

**Expected behavior**  
Correct display of parameters(load, temp, memory) of Intel integrated UHD 770 GPU

**Screenshots / photos of the Turing screen**  
[image]

[image]

**Environment:**  
 - Smart screen model Turing 5"
 - main.py
 - OS Windows 10
 - Python 3.10.2
 - Hardware Intel CPU I5-3600K,  Intel integrated UHD 770 GPU

**Additional context**  

**Report of test_librehardwaremonitor.py**

---------------------------------------------------------------------------------------
= RESTART: C:\Program Files\Python310\turing-smart-screen-python-main\external\LibreHardwareMonitor\test_librehardwaremonitor.py
Found LibreHardwareMonitorLib 0.9.3.0
Found HidSharp 2.1.0.0
Motherboard | Gigabyte B760I AORUS PRO | /motherboard
    SuperIO | ITE IT8689E | /lpc/it8689e/0
        Control | Fan #1 | None
        Control | Fan #2 | None
        Control | Fan #3 | 50.0
        Control | Fan #4 | None
        Control | Fan #5 | None
        Control | Fan #6 | None
        Voltage | Voltage #1 | 0.7319999933242798
        Voltage | Voltage #2 | 2.0160000324249268
        Voltage | Voltage #3 | 2.003999948501587
        Voltage | Voltage #4 | 2.003999948501587
        Voltage | Voltage #5 | 0.035999998450279236
        Voltage | Voltage #6 | 1.8000000715255737
        Voltage | Voltage #7 | 1.9919999837875366
        Voltage | 3VSB | 3.3359999656677246
        Voltage | VBat | 3.0480000972747803
        Temperature | Temperature #1 | 39.0
        Temperature | Temperature #2 | 40.0
        Temperature | Temperature #3 | 42.0
        Temperature | Temperature #5 | 39.0
        Fan | Fan #1 | 470.3832702636719
        Fan | Fan #2 | 0.0
        Fan | Fan #5 | 5625.0
----------------------------------------------------
Cpu | 13th Gen Intel Core i5-13600K | /intelcpu/0
    Load | CPU Core #1 Thread #1 | 7.540840148925781
    Load | CPU Core #1 Thread #2 | 0.19467473030090332
    Load | CPU Core #2 Thread #1 | 0.5620360374450684
    Load | CPU Core #2 Thread #2 | 0.19320249557495117
    Load | CPU Core #3 Thread #1 | 0.0
    Load | CPU Core #3 Thread #2 | 0.18829107284545898
    Load | CPU Core #4 Thread #1 | 0.0
    Load | CPU Core #4 Thread #2 | 0.06903409957885742
    Load | CPU Core #5 Thread #1 | 0.0
    Load | CPU Core #5 Thread #2 | 0.436168909072876
    Load | CPU Core #6 Thread #1 | 0.0
    Load | CPU Core #6 Thread #2 | 0.43587684631347656
    Load | CPU Core #7 | 5.169338226318359
[... 196 lines cut ...]
23.02.2024 11:59:10 [DEBUG] Drawing Text: WLDOWN
23.02.2024 11:59:10 [DEBUG] Drawing Text: WLUP
23.02.2024 11:59:10 [DEBUG] Drawing Text: DISKLOAD
23.02.2024 11:59:10 [DEBUG] Drawing Text: DISKFREE
23.02.2024 11:59:10 [DEBUG] Found one supported GPU: Intel(R) UHD Graphics 770
Exception in thread CPU_FanSpeed:
Traceback (most recent call last):
23.02.2024 11:59:10 [WARNING] Your GPU memory absolute usage (M) is not supported yet
  File "C:\Program Files\Python310\lib\threading.py", line 1009, in _bootstrap_inner
23.02.2024 11:59:10 [WARNING] Your GPU load is not supported yet
    self.run()23.02.2024 11:59:10 [WARNING] Your GPU temperature is not supported yet

  File "C:\Program Files\Python310\lib\threading.py", line 946, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Program Files\Python310\turing-smart-screen-python-main\library\scheduler.py", line 73, in wrap
    periodic(scheduler, interval, func)
  File "C:\Program Files\Python310\turing-smart-screen-python-main\library\scheduler.py", line 64, in periodic
    action(*actionargs)
  File "C:\Program Files\Python310\turing-smart-screen-python-main\library\scheduler.py", line 118, in CPUFanSpeed
    stats.CPU.fan_speed()
  File "C:\Program Files\Python310\turing-smart-screen-python-main\library\stats.py", line 307, in fan_speed
    fan_percent = sensors.Cpu.fan_percent()
  File "C:\Program Files\Python310\turing-smart-screen-python-main\library\sensors\sensors_librehardwaremonitor.py", line 238, in fan_percent
    return float(sensor.Value)
TypeError: float() argument must be a string or a real number, not 'NoneType

mathoudebine, 2024-02-25:
Hi, thanks for raising this issue. I do not have an Intel GPU at home so couldn't test it.
Your GPU offers these sensors through LibreHardwareMonitor:
```
----------------------------------------------------
GpuIntel | Intel(R) UHD Graphics 770 | /gpu-intel-integrated/\\?\PCI#VEN_8086&DEV_A780&SUBSYS_D0001458&REV_04#3&11583659&0&10#{1ca05180-a699-450a-9a0c-de4fbe3ddd89}
    Power | GPU Power | 12.611653327941895
    SmallData | D3D Shared Memory Used | 605.921875
    Load | D3D 3D | 4.135100364685059
    Load | D3D Video Decode | 0.0
    Load | D3D Copy | 0.0
    Load | D3D Video Processing | 0.0
    Load | D3D Video Decode | 0.0
    Load | D3D Other | 0.0
    Load | D3D Other | 0.0
    Load | D3D Other | 0.0
    Load | D3D Other | 0.0
    Load | D3D Other | 0.0
    Load | D3D GDI Render | 0.0
    Load | D3D Video Processing | 0.0
    Load | D3D Other | 0.0
```
Usually AMD/Nvidia GPUs have a "GPU Core" sensor for GPU global usage. Here it is not available for your Intel GPU but the "D3D 3D" sensor can be used instead. It will be less precise than a "GPU Core" sensor but this is all we have.

For the memory usage the "D3D Shared Memory Used" can be used.

Unfortunately there is no temperature sensor available for your GPU.

I will make a pull request for the GPU usage and GPU memory usage to fix it.

mathoudebine, 2024-02-25:
The issue should be fixed on `main` branch! Can you give it a try?

mathoudebine, 2024-02-26:
You just need to download the project again from the main branch to get the latest changes: you can do this from the Github homepage or using this direct link: https://github.com/mathoudebine/turing-smart-screen-python/archive/refs/heads/main.zip

ingenegra, 2024-02-27:
Hi! I reload files from Github. I took screenshots of the windows10 utility, intel utility and test_librehardwaremonitor. The 5" display showed the download corresponding to the number displayed in the Windows 10 utility in the 3d window. But the GPU load is considered differently. Screenshots attach -- С уважениемВячеслав Атрощенко