mathoudebine/turing-smart-screen-python #478: Another new 3.5" screen?

braewoods, 2024-02-24:
Found another one that appears to have different proprietary software than others: 
https://www.aliexpress.com/item/3256805998169670.html

mathoudebine, 2024-02-24:
Thanks for the link! I've bought one and I guess we'll see if it can be integrated or not

siffland, 2024-06-29:
I think i have the same model as this one

https://www.aliexpress.us/item/3256806436893410.html?spm=a2g0o.order_list.order_list_main.17.737f1c24iUhrPn&gatewayAdapt=glo2usa

If i use type A, it tries to load, but I just get the default screen saying

"GUITION MOnitor Screen"

and the python code hangs

```
me@server:/turing-smart-screen-python-3.5.0$ python3 main.py
06/29/24 01:55:16 [INFO] Loading theme LandscapeMagicBlue from res/themes/LandscapeMagicBlue/theme.yaml
06/29/24 01:55:16 [DEBUG] HW revision: A
06/29/24 01:55:16 [DEBUG] Static COM port: /dev/ttyACM0
06/29/24 01:55:16 [DEBUG] Using Python 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0]
06/29/24 01:55:16 [WARNING] Tray icon is not supported on your platform
06/29/24 01:55:16 [INFO] Display reset (COM port may change)...
06/29/24 01:55:21 [DEBUG] Static COM port: /dev/ttyACM0
06/29/24 01:55:21 [DEBUG] HW sub-revision: SubRevision.TURING_3_5
06/29/24 01:55:21 [DEBUG] Drawing Image: BACKGROUND
06/29/24 01:55:21 [DEBUG] Bitmap res/themes/LandscapeMagicBlue/background.png is now loaded in the cache
06/29/24 01:55:21 [DEBUG] Drawing Text: DISK_LABEL
06/29/24 01:55:21 [DEBUG] Drawing Text: DISK_USED_LABEL
06/29/24 01:55:21 [INFO] Detected Nvidia GPU(s)
```

I am using ubuntu 22.04 (no GUI, just CLI).  Any chance of getting it working.

mathoudebine, 2024-06-30:
@siffland It seems you have the Guition Smart screen, unfortunately it is not supported by this project, see here: https://github.com/mathoudebine/turing-smart-screen-python/issues/426