mathoudebine/turing-smart-screen-python #489: Wrong CPU and GPU model, wrong time in wallclock time, etc.

fbreve, 2024-03-23:
I'm using the TaskManager5inch theme and I found some issues:
- The CPU model is wrong: it is showing "Core I9 I10900K" but my processor is a Core i9 14900K.
- The GPU model is wrong: it is showing "RTX3080TI" but my GPUs are a RTX 4060 Ti and a RTX 2060 SUPER.
- When the computer enters the sleep state, the screen stays on and the wallclock time doesn't update anymore. After the system is resumed the clock goes on from where it stops, therefore it's behind the actual time.
- The date in Portuguese show as "23 de mar. de 2024". This is correct, but it is a long string, which truncates with the "Clock" text.

**Environment:**  
 - Smart screen model: UsbPCMonitor 5"
 - Revision of this project: 3.4.0
 - OS with version: Windows 11
 - Python version: 3.11
 - Hardware: Core i9 14900K processor, Nvidia GPUs RTX 4060 Ti and RTX 2060 SUPER

DJRoby19, 2024-03-25:
CPU model and GPU model are fixed variables that you can change inside the theme.yaml file in turing-smart-screen-python/res/themes/TaskManager5inch
on line 19 and 27.

fbreve, 2024-03-26:
Thanks, I edited the theme to match those texts with my hardware.

I found more issues: the wall clock seems to stay behind even if the computer is on. The stats also doesn't match what is going on, maybe they are delayed too.

I'm trying Landscape15grid now to see if the issue is only related to the TaskManager theme.

mathoudebine, 2024-04-26:
Nice to know you managed to fix your issue! I'm closing this ticket, but feel free to open a new issue in case you have more problems with the program

closed 2024-04-26 by mathoudebine