mathoudebine/turing-smart-screen-python #493: 5 inch display not work when i mount the HDD to virtual machine

vinprom1, 2024-03-29:
I use USBPCMonitorENG and connect the 5 inch display to it, which starts and works normally, but if I start a virtual linux machine with vmware workstation, to which I have mounted a physical hard disk (then the hdd becomes inactive and cannot be accessed in windows, it is occupied by linux) and the program closes itself and does not start.
Here are the windows log files:

Faulting application name: UsbPCMonitor.exe, version: 1.0.0.0, time stamp: 0x64eee5ec
Faulting module name: KERNELBASE.dll, version: 10.0.19041.4170, time stamp: 0x96969434
Exception code: 0xe0434352
Fault offset: 0x0013fa72
Faulting process id: 0x103c
Faulting application start time: 0x01da81ed3f89311f
Faulting application path: C:\USBPCMonitorENG\UsbPCMonitor.exe
Faulting module path: C:\Windows\System32\KERNELBASE.dll
Report Id: 63cb3fc3-6c0c-4b83-ba6c-79fdc77dcbb5
Faulting package full name: 
Faulting package-relative application ID:

mathoudebine, 2024-04-26:
This project `turing-smart-screen-python` on Gihtub is not the same program as `UsbPCMonitor.exe`
Please read disclaimer from the front page here: https://github.com/mathoudebine/turing-smart-screen-python?tab=readme-ov-file#%EF%B8%8F-disclaimer---please-read-%EF%B8%8F

UsbPcMonitor has been developed by the screen manufacturer, you should ask the seller of your screen if they have a new version that fixes your issue

closed 2024-04-26 by mathoudebine

vinprom1, 2024-04-26:
;( ok thanks