mathoudebine/turing-smart-screen-python #498: Intel i7-12700 no data showing up

pac2003man, 2024-04-10:
**Describe the bug**  
CPU Data Only Shows Usage

**To Reproduce**  
Steps to reproduce the behavior:
On Startup

**Expected behavior**  
Show CPU Data

**Screenshots / photos of the Turing screen**  
[image]

**Environment:**  
 - Smart screen model: XuanFang 3.5
 - Revision of this project 
 - OS with version Windows 11 23H2
 - Python version 3.10.11
 - Hardware 12th Gen Intel Core i7-12700

**Additional context**  
Main Log + test_librehardwaremonitor.py Log below

Program is runnig as Admin!

`
File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\threading.py", line 1016, in _bootstrap_inner
10.04.2024 15:22:44 [WARNING] Found 2 GPUs on your system (0 AMD / 1 Nvidia / 1 Intel). Auto identify which GPU to use.
    self.run()
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.3056.0_x64__qbz5n2kfra8p0\lib\threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\Besitzer\Documents\miniscreen\library\scheduler.py", line 73, in wrap
    periodic(scheduler, interval, func)
  File "C:\Users\Besitzer\Documents\miniscreen\library\scheduler.py", line 64, in periodic
    action(*actionargs)
  File "C:\Users\Besitzer\Documents\miniscreen\library\scheduler.py", line 110, in CPUTemperature
    stats.CPU.temperature()
  File "C:\Users\Besitzer\Documents\miniscreen\library\stats.py", line 280, in temperature
    temperature = sensors.Cpu.temperature()
  File "C:\Users\Besitzer\Documents\miniscreen\library\sensors\sensors_librehardwaremonitor.py", line 214, in temperature
    return float(sensor.Value)
TypeError: float() argument must be a string or a real number, not 'NoneType'
10.04.2024 15:22:44 [DEBUG] This GPU will be used for stats: NVIDIA GeForce RTX 3070

Found LibreHardwareMonitorLib 0.9.3.0
Found HidSharp 2.1.0.0
Motherboard | Gigabyte B660M DS3H AX DDR4 | /motherboard
----------------------------------------------------
Cpu | 12th Gen Intel Core i7-12700 | /intelcpu/0
    Load | CPU Core #1 Thread #1 | 13.587522506713867
    Load | CPU Core #1 Thread #2 | 0.38230419158935547
    Load | CPU Core #2 Thread #1 | 4.559493064880371
    Load | CPU Core #2 Thread #2 | 0.0
    Load | CPU Core #3 Thread #1 | 0.0
    Load | CPU Core #3 Thread #2 | 0.0
    Load | CPU Core #4 Thread #1 | 0.0
    Load | CPU Core #4 Thread #2 | 0.4942655563354492
    Load | CPU Core #5 Thread #1 | 1.11808180809021
    Load | CPU Core #5 Thread #2 | 24.07919692993164
    Load | CPU Core #6 Thread #1 | 36.152503967285156
    Load | CPU Core #6 Thread #2 | 1.8293440341949463
    Load | CPU Core #7 Thread #1 | 0.49278736114501953
    Load | CPU Core #7 Thread #2 | 21.897720336914062
    Load | CPU Core #8 Thread #1 | 7.418745994567871
    Load | CPU Core #8 Thread #2 | 0.34897327423095703
    Load | CPU Core #9 | 17.17165756225586
    Load | CPU Core #10 | 16.50577163696289
    Load | CPU Core #11 | 17.42238998413086
    Load | CPU Core #12 | 16.865562438964844
    Load | CPU Total | 8.737259864807129
[... 155 lines cut ...]
    Load | Network Utilization | 0.0
----------------------------------------------------
Network | vEthernet (BluestacksNxt) | /nic/{DD652D93-D4FD-4324-850E-15D725CDDEAB}
    Data | Data Uploaded | 28.21662139892578
    Data | Data Downloaded | 0.461410790681839
    Throughput | Upload Speed | 0.0
    Throughput | Download Speed | 0.0
    Load | Network Utilization | 0.0
----------------------------------------------------
Network | vEthernet (Default Switch) | /nic/{7FB9E47A-E931-4222-8BDA-CE7393135775}
    Data | Data Uploaded | 0.014429082162678242
    Data | Data Downloaded | 0.0
    Throughput | Upload Speed | 0.0
    Throughput | Download Speed | 0.0
    Load | Network Utilization | 0.0
----------------------------------------------------
Network | WLAN | /nic/{CCF45E6E-6594-45A4-A8E2-B66946BFC166}
    Data | Data Uploaded | 0.0
    Data | Data Downloaded | 0.0
    Throughput | Upload Speed | 0.0
    Throughput | Download Speed | 0.0
    Load | Network Utilization | 0.0
----------------------------------------------------

`

mathoudebine, 2024-04-26:
Hi, it seems temperature and clock data from your CPU are not read properly by LibreHardwareMonitor
Can you download the LHM full program from here: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/releases/download/v0.9.3/LibreHardwareMonitor-net472.zip and run it?
Can you see temperature data from your CPU like in this screenshot?
[image]

closed 2024-06-18 by mathoudebine

GuyFromTurkey, 2024-08-17:
@mathoudebine i don't have CPU temperature when i open LibreHardwareMonitor.exe

[image]

GuyFromTurkey, 2024-08-17:
> @mathoudebine i don't have CPU temperature when i open LibreHardwareMonitor.exe [...]

My antivirus was triggered when I ran test_librehardwaremonitor.py "LibreHardwareMonitor.sys" was potentially dangerous, I added an exception and I had my values ​​from my CPU.
I add exception too with "python.sys" who is at this path "C:\Users\xxx\AppData\Local\Programs\Python\Python312"

Everything works now, I hope I was able to help you.