mathoudebine/turing-smart-screen-python #499: Distorted values when running on macOS

JsThiago, 2024-04-15:
**Describe the bug**  
Hi, When I use the screen on macOS, some values become distorted or does not appear on the screen. I tried the same cable but on a Linux machine and worked fine. I also tested USB port with an external drive and I get around 450MB/s read and write.

**To Reproduce**  
Steps to reproduce the behavior:
1. Download the 3.4.0 release
2. Connect the screen at the USB port
3. Run pip3 install -r requirements.txt
4. Run python3 main.py
5. See some values distorted or not appearing on the screen.

**Expected behavior**  
The expected behavior is the values appearing on the screen and without being distorted.

**Screenshots / photos of the Turing screen**  

[image]

**Environment:**  
 - Screen: "UsbPCMonitor" 3.5"
 - Code version: Release version 3.4.0 of this application
 - OS version: macOS Sonoma 14.4
 - Python version: Python 3.11
 - Machine: Mac mini m2 2023

**Additional context**  
The code worked fine using a machine running Rock Linux. I was using the same cable and a USB 3.0 port. None distorted image show and all values appeared on the screen.

c0ldJS, 2024-04-15:
@JsThiago : Might be interested in issue #7 and some of the tips/findings there.

JsThiago, 2024-04-16:
> @JsThiago : Might be interested in issue #7 and some of the tips/findings there.

Thanks. I was able to use the fix/replace-queued-data-send-with-immediate-plus-deferred-delay branch. But, as this branch is 2 years old, it wasn't compiling.  I did something that I do not remember and was able to compile but, something did not work like the landscape (Yeah it only works in portrait mode, not even reverse landscape), some worked like the charts. But for me is better than what was before. Thanks for your help :)

closed 2024-04-16 by JsThiago