mathoudebine/turing-smart-screen-python #504: Xuanfang Rev b/ flagship isnt working

WastelandwalkerYT, 2024-04-25:
**Describe the bug**  
I bought the 2.5 inch. Revision b with the l.eD backing And I tried the stock firmware. It didn't work, so I tried the python code. And I am experiencing the same problem. The screen is stuck on the stock screen as shown in pictures

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to open the python app
2. Click on save and start
3. Cmd lines says it's working and tray is opened but no screen change on 3.5 inch

**Expected behavior**  
I expected it to work XD

**Screenshots / photos of the Turing screen**  
Add screenshots or photos of the rendering on the smart screen to help explain your problem.
You can drag and drop photos here to add them to the description.

**Environment:**  
 - Smart screen model xuanfang Rev b and flagship 3.5 inch
 - Revision of this project main branch newest version
 - OS with version windows 10
 - Python version  Python 3.12
 - Hardware cpu intel core i5-7300HQ gpus Nvidiagforcegtx 1050 & intel hd grafics 630

**Additional context**  
Not sure whats wrong here as it is Recognized under static com port 10 and is working as far as the program is Concerned.
[image]
[image]
[image]

mathoudebine, 2024-04-26:
Hi,
It looks like you do not have a Xuanfang smart screen. Xuanfang screens do not have default content displayed when plugged, they stay black.
You may have another smart screen model like the Guition or the Fuldho smart screen.
This also explain why your screen is not working with default software.

Unfortunately, your screen is not supported by this project: only screen listed on the frontpage here are supported: https://github.com/mathoudebine/turing-smart-screen-python?tab=readme-ov-file#supported-smart-screens-models
You should ask the seller for the official software that works with your screen

WastelandwalkerYT, 2024-04-28:
That's extremely strange! They already have bootleg versions it seems XD thanks for the info I've contacted the seller on Amazon. Waiting for reply but in the mean time here's some more pictures of the device and box 
[image]
[image]
[image]
[image]

mathoudebine, 2024-04-29:
I agree, the backplate seems very close to the Xuanfang display!
Please keep me informed about what the seller will tell you, I'm interested if there is a link to a software to download

WastelandwalkerYT, 2024-04-30:
So far still waiting on seller to message me but I did take the backing off of the device. Found out there's no led lights installed. Leading me to believe this is a bootleg version. Attached are some pics of the board
[image]
[image]
[image]
[image]

zurailuridze, 2024-10-03:
Maybe add this model and software to the main thread. I have the software for it, but I don't fully understand how it works.

zurailuridze, 2024-10-03:
[image]

towlerj, 2025-03-09:
I have the same model - what would be the process for adding a new screen?

I assume step 1. is adding an entry to turing-smart-screen-python\tools\lsusb for it. I'm currently running Windows, but I'll see if I can generate something.

Rip-07, 2025-04-26:
Mb you can use this software
https://smartdisplay.lanzouo.com/b04lebxba

Rip-07, 2025-04-26:
Also I found this 

[image]

towlerj, 2025-04-27:
Thanks, I ended up ditching it as it just stopped working entirely.