mathoudebine/turing-smart-screen-python #512: Commands or parameters "missed" with Apparent UsbPCMonitor 3.5"

jeffsf, 2024-05-07:
A 3.5" screen was purchased on AliExpress that was indicated in the comments to work with this software. Running `main.py` results in some areas "garbled" after refreshes. Running a relatively simple program to write a black image and then lines of text sometimes does not write the black image and results in misplaced or garbled text (upper part of the text shifted relative to the lower part). 

**To Reproduce**

_Using `main.py`_

- Plug in display to hub on Mac.
- Run `main.py`
```
05/07/2024 10:26:39 [INFO] Loading theme 3.5inchTheme2 from res/themes/3.5inchTheme2/theme.yaml
05/07/2024 10:26:39 [DEBUG] HW revision: A
05/07/2024 10:26:39 [DEBUG] Auto detected COM port: /dev/cu.usbmodemUSB35INCHIPSV21
```
- Note that areas are "random" dots rather than text or clean graphical elements
- Note that some of these areas resolve with apparent refreshes and others become garbled

_Using custom code_

- Run the attached code, `scrolling-text.py` from the root level of this project's local directory
- Observe that the screen does not go black consistently 
- Observe that the text is garbled and may be in the wrong "row"
- Note that on subsequent runs, the resulting display is dfifferent

Neither setting `update_queue=None` nor adding `time.sleep()` between calls seemed to resolve the rendering issues.

**Expected behavior**  

The screen would be refreshed without significant rendering defects

**Environment:**  
 - Apparent UsbPCMonitor 3.5"
   - "Choice SKU 759133297043"
   - AliExpress 3256806256906629.html
   - "New 3.5 Inch IPS TYPE-C Secondary Screen Computer CPU GPU RAM HDD MonitorUSB Display For Freely AIDA64"
 - Commit 2355937 on `main`
 - macOS Sonoma 14.4.1 on Mac Mini 2018 (3.2 GHz 6-Core Intel Core i7)
 - Python 3.12.3

**Desired Outcome:**

Mitigation approach or reasonable confidence in an assessment that this display is a PoS

[image]
[image]
[image]

jeffsf, 2024-05-07:
```
#!/usr/bin/env python

# Program to see how fast the "New 3.5 Inch IPS TYPE-C Secondary Screen ..."
# `USB35INCHIPSV21` can scroll text.
#
# Appears to be a 480x320 device.
# 8x16 characters would be 20 rows of 60 characters

import logging
import time

from library.lcd.lcd_comm import Orientation
from library.lcd.lcd_comm_rev_a import LcdCommRevA, Command
from library.log import logger

# Modify the logging format
new_formatter = logging.Formatter(
    fmt='%(asctime)s,%(msecs)03d %(levelname)s: %(message)s',
    datefmt='%Y-%m-%d %H:%M:%S',
)
for handler in logger.parent.handlers:
    handler.setFormatter(new_formatter)

# dimensions are when in portrait
# Setting update_queue=None from lcd_comm.py:
#     If you want serial requests to be done in sequence, set it to None
lcd_comm = LcdCommRevA(com_port="AUTO",
                       display_width=320,
                       display_height=480,
                       update_queue=None)

lcd_comm.Reset()
lcd_comm.InitializeComm()

lcd_comm.SetBrightness(25)
lcd_comm.SetOrientation(orientation=Orientation.LANDSCAPE)

from library.log import logger

# lcd_comm.SendCommand(Command.TO_BLACK, 0, 0, 320, 320)  # No impact

lcd_comm.DisplayBitmap("./480x320_black.png")

logger.info("Black written, sleeping 2")
time.sleep(2)

logger.info("Writing text")

for row in range(0,9):
    offset = row * 16
    text = (f"{row+1:02x}-4567891"
            "1234567892"
            "1234567893"
            "1234567894"

    )
    lcd_comm.DisplayText(
        text=text,
        x=0, y=offset,
        font="roboto-mono/RobotoMono-Regular.ttf",
        font_size=16,
        font_color=(255, 255, 255),
        background_color=(10, 10, 10))
    time.sleep(1)

logger.info("Writing text done")

time.sleep(2)
```

jeffsf, 2024-05-07:
Image of box label and insert

[image]

[05072024_3.5 Inch IPS TYPE-C Secondary Screen Instructions 1. Do.pdf](https://github.com/mathoudebine/turing-smart-screen-python/files/15239164/05072024_3.5.Inch.IPS.TYPE-C.Secondary.Screen.Instructions.1.Do.pdf)

jeffsf, 2024-05-07:
Both scripts work on Debian Bookworm.

Updating title to indicate that this appears to be a macOS issue

c0ldJS, 2024-05-08:
Have you taken a look at issues #7 and #499 ?

Numbski, 2024-06-12:
I am getting this exact same behavior.