mathoudebine/turing-smart-screen-python #514: journalctl reports No module named GPUtil

pakair, 2024-05-14:
/home/user/turing/python3 main.py works. 
That tells me all modules are loaded in python3. 
My login is user.

moved /home/user/turing/ to /opt
copied .service to /etc/systemd/system/
sudo systemctl daemon-reload

sudo systemctl start turing.service fails

journalctl -u turing.service says No module GPUtil

Linuxmint 21.3, python3.10.12, turing display 3.5 inch

pakair, 2024-05-14:
```
user@cw:/opt/turing-smart-screen-python$ pip install -r requirements.txt

Defaulting to user installation because normal site-packages is not writeable
Ignoring numpy: markers 'python_version < "3.9"' don't match your environment
Ignoring GPUtil: markers 'python_version >= "3.12"' don't match your environment
Ignoring pyadl: markers 'sys_platform == "win32"' don't match your environment
Ignoring pythonnet: markers 'sys_platform == "win32"' don't match your environment
Ignoring pywin32: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: Pillow~=10.2.0 in /home/user/.local/lib/python3.10/site-packages (from -r requirements.txt (line 2)) (10.2.0)
Requirement already satisfied: pyserial~=3.5 in /usr/lib/python3/dist-packages (from -r requirements.txt (line 3)) (3.5)
Requirement already satisfied: PyYAML~=6.0.1 in /home/user/.local/lib/python3.10/site-packages (from -r requirements.txt (line 4)) (6.0.1)
Requirement already satisfied: psutil~=5.9.8 in /home/user/.local/lib/python3.10/site-packages (from -r requirements.txt (line 5)) (5.9.8)
Requirement already satisfied: pystray~=0.19.5 in /home/user/.local/lib/python3.10/site-packages (from -r requirements.txt (line 6)) (0.19.5)
Requirement already satisfied: babel~=2.14.0 in /home/user/.local/lib/python3.10/site-packages (from -r requirements.txt (line 7)) (2.14.0)
Requirement already satisfied: ruamel.yaml~=0.18.6 in /home/user/.local/lib/python3.10/site-packages (from -r requirements.txt (line 8)) (0.18.6)
Requirement already satisfied: sv-ttk~=2.6.0 in /home/user/.local/lib/python3.10/site-packages (from -r requirements.txt (line 9)) (2.6.0)
Requirement already satisfied: numpy~=1.26.4 in /home/user/.local/lib/python3.10/site-packages (from -r requirements.txt (line 13)) (1.26.4)
Requirement already satisfied: GPUtil~=1.4.0 in /home/user/.local/lib/python3.10/site-packages (from -r requirements.txt (line 16)) (1.4.0)
Requirement already satisfied: pyamdgpuinfo~=2.1.6 in /home/user/.local/lib/python3.10/site-packages (from -r requirements.txt (line 21)) (2.1.6)
Requirement already satisfied: python-xlib>=0.17 in /usr/lib/python3/dist-packages (from pystray~=0.19.5->-r requirements.txt (line 6)) (0.29)
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from pystray~=0.19.5->-r requirements.txt (line 6)) (1.16.0)
Requirement already satisfied: ruamel.yaml.clib>=0.2.7 in /home/user/.local/lib/python3.10/site-packages (from ruamel.yaml~=0.18.6->-r requirements.txt (line 8)) (0.2.8)

```

pakair, 2024-05-14:
I had to change User=root to **User=user** in the service file. (**user** is my login)

I am able to start/stop the service and the display responds.

mathoudebine, 2024-06-18:
Hi, thanks for your feedback!
There is indeed an issue with the service file, it should not run the program as root because the Python dependencies are installed locally for the user. I will fix the issue in the file

closed 2024-06-18 by mathoudebine