mathoudebine/turing-smart-screen-python #516: On mac, python-tk is already installed via homebrew, but the error to install tkinter persists.

Numbski, 2024-06-04:
**Describe the bug**  
```Garou:~/git/turing-smart-screen-python  tshadwick$ brew install python-tk
==> Downloading https://formulae.brew.sh/api/formula.jws.json
##O=#    #
==> Downloading https://formulae.brew.sh/api/cask.jws.json

Warning: python-tk@3.12 3.12.3 is already installed and up-to-date.
To reinstall 3.12.3, run:
  brew reinstall python-tk@3.12
Garou:~/git/turing-smart-screen-python  tshadwick$ python3 configure.py
[ERROR] Tkinter dependency not installed. Please follow troubleshooting page: https://github.com/mathoudebine/turing-smart-screen-python/wiki/Troubleshooting#all-os-tkinter-dependency-not-installed
Garou:~/git/turing-smart-screen-python  tshadwick$```

**To Reproduce**  
Steps to reproduce the behavior:
1. ```homebrew update && homebrew upgrade && homebrew install python-tk``
2. ```python3 configure.py```

**Expected behavior**  
Literally anything but this error.

**Screenshots / photos of the Turing screen**  
Add screenshots or photos of the rendering on the smart screen to help explain your problem.
You can drag and drop photos here to add them to the description.

**Environment:**  
 - Smart screen model [e.g. Turing 5", XuanFang 3.5”]
 - Revision of this project [e.g. 3.0.0, `main` branch, specific commit]
 - OS with version [e.g. Windows 11, Ubuntu 22.04]
 - Python version [e.g. Python 3.10]
 - Hardware [e.g. Intel CPU, Nvidia GPU, Raspberry Pi 3 B+ ...]

**Additional context**  
Add any other context about the problem here.

Numbski, 2024-06-12:
This is so weird.  I switched to a venv and this resolved itself.  So very confusing.

closed 2024-06-12 by Numbski

0xnth, 2024-10-12:
> This is so weird.  I switched to a venv and this resolved itself.  So very confusing.
Yoo I have the same problem as you but no way to succeed on macos sonoma, my panel is smart screen turning TURZX 3.5", i will try installing python with venv