mathoudebine/turing-smart-screen-python #537: Cannot select GPU to use when there's an APU/iGPU available

Virkkunen, 2024-07-15:
**Describe the bug**  
I have an AMD 7800X3D and a NVIDIA RTX 3080. Previously I had an AMD 5900X, which does not have an iGPU, so the script correctly identified and used my 3080. Now that I've changed the CPU and the 7800X3D has an iGPU, it's always detected and prioritised over my NVIDIA card.

**Environment:**  
 - Smart screen model: Turing 3.5"
 - Revision of this project: commit 67661cba71d8531d3682f2a5ba209b8e3ff650e1
 - OS with version: endeavourOS
 - Python version: Python 3.12.4
 - Hardware: AMD Ryzen 7 7800X3D, NVIDIA RTX 3080

**Additional context**  
The logs only show this:
```
5 20:59:46 endeavour python[34570]: 15/07/24 20:59:46 [DEBUG] HW sub-revision: SubRevision.USBMONITOR_3_5
Jul 15 20:59:46 endeavour python[34570]: 15/07/24 20:59:46 [DEBUG] Drawing Image: BACKGROUND
Jul 15 20:59:46 endeavour python[34570]: 15/07/24 20:59:46 [DEBUG] Bitmap res/themes/Catpunk/background.png is now loaded in the cache
Jul 15 20:59:46 endeavour python[34570]: 15/07/24 20:59:46 [DEBUG] Drawing Text: CPU_MODEL
Jul 15 20:59:46 endeavour python[34570]: 15/07/24 20:59:46 [DEBUG] Drawing Text: GPU_MODEL
Jul 15 20:59:46 endeavour python[34570]: 15/07/24 20:59:46 [DEBUG] Drawing Text: RAM_MODEL
Jul 15 20:59:46 endeavour python[34570]: 15/07/24 20:59:46 [DEBUG] Drawing Text: DOWNLOAD_RATE
Jul 15 20:59:46 endeavour python[34570]: 15/07/24 20:59:46 [DEBUG] Drawing Text: UPLOAD_RATE
Jul 15 20:59:46 endeavour python[34570]: 15/07/24 20:59:46 [DEBUG] Drawing Text: DISK_USAGE
Jul 15 20:59:46 endeavour python[34570]: 15/07/24 20:59:46 [INFO] Detected AMD GPU(s)
```

Virkkunen, 2024-07-16:
On context, this works fine on my Windows installation, it correctly shows the NVIDIA card. The logs:
```
16/07/2024 13:17:10 [DEBUG] HW revision: A
16/07/2024 13:17:10 [DEBUG] Auto detected COM port: COM6
16/07/2024 13:17:10 [DEBUG] Found LibreHardwareMonitorLib 0.9.3.0
16/07/2024 13:17:12 [INFO] Found CPU: AMD Ryzen 7 7800X3D
16/07/2024 13:17:12 [INFO] Found Memory: Generic Memory
16/07/2024 13:17:12 [INFO] Found AMD GPU: AMD Radeon(TM) Graphics
16/07/2024 13:17:12 [INFO] Found Nvidia GPU: NVIDIA GeForce RTX 3080
16/07/2024 13:17:12 [INFO] Found Storage: CT1000MX500SSD1
16/07/2024 13:17:12 [INFO] Found Storage: WDS100T3X0C-00SJG0
16/07/2024 13:17:12 [INFO] Found Storage: WD_BLACK SN850X HS 1000GB
16/07/2024 13:17:12 [INFO] Found Network interface: Bluetooth Network Connection 5
16/07/2024 13:17:12 [INFO] Found Network interface: Ethernet
16/07/2024 13:17:12 [INFO] Found Network interface: Local Area Connection* 10
16/07/2024 13:17:12 [INFO] Found Network interface: Local Area Connection* 11
16/07/2024 13:17:12 [INFO] Found Network interface: WiFi
16/07/2024 13:17:12 [DEBUG] Using Python 3.11.3 (tags/v3.11.3:f3909b8, Apr  4 2023, 23:49:59) [MSC v.1934 64 bit (AMD64)]
16/07/2024 13:17:12 [INFO] Tray icon has been displayed
16/07/2024 13:17:12 [INFO] Display reset (COM port may change)...
16/07/2024 13:17:17 [DEBUG] Static COM port: COM6
16/07/2024 13:17:17 [DEBUG] HW sub-revision: SubRevision.USBMONITOR_3_5
16/07/2024 13:17:17 [DEBUG] Drawing Image: BACKGROUND
16/07/2024 13:17:17 [DEBUG] Bitmap res/themes/Catpunk/background.png is now loaded in the cache
16/07/2024 13:17:17 [DEBUG] Drawing Text: CPU_MODEL
16/07/2024 13:17:17 [DEBUG] Drawing Text: GPU_MODEL
16/07/2024 13:17:17 [DEBUG] Drawing Text: RAM_MODEL
16/07/2024 13:17:17 [DEBUG] Drawing Text: DOWNLOAD_RATE
16/07/2024 13:17:17 [DEBUG] Drawing Text: UPLOAD_RATE
16/07/2024 13:17:17 [DEBUG] Drawing Text: DISK_USAGE
16/07/2024 13:17:17 [WARNING] Found 2 GPUs on your system (1 AMD / 1 Nvidia / 0 Intel). Auto identify which GPU to use.
16/07/2024 13:17:17 [DEBUG] This GPU will be used for stats: NVIDIA GeForce RTX 3080
```

Virkkunen, 2024-08-12:
I didn't see this mentioned anywhere but branch https://github.com/mathoudebine/turing-smart-screen-python/tree/fix/537-cannot-select-gpu-to-use-when-theres-an-apuigpu-available fixes this issue. Apparently it was a very easy solution to just change the priorisation under the is_Available function

mathoudebine, 2024-09-09:
Hello @Virkkunen indeed I made a branch with the fix but didn't have the time to merge it. It is now done, thank you for your feedback!

PR #571 (2024-09-09): Use Nvidia GPU in priority with Python sensors
Closes #537
closed 2024-09-09 by mathoudebine