mathoudebine/turing-smart-screen-python #545: [Feature Request] Option to select GPU

fbreve, 2024-07-29:
I currently have 2 discrete GPUs on my system, the main one is an RTX 4060 Ti and the secondary one is an RTX 2060 SUPER. I want to display the data from the main one, as the other remains idle most of the time. However, there the secondary is auto-selected:

29/07/2024 12:56:37 [WARNING] Found 3 GPUs on your system (0 AMD / 2 Nvidia / 1 Intel). Auto identify which GPU to use.
29/07/2024 12:56:37 [DEBUG] This GPU will be used for stats: NVIDIA GeForce RTX 2060 SUPER

Virkkunen, 2024-08-11:
On extension, this is also related to #537 where I can't select between my iGPU and dGPU as it automatically selects my iGPU.

ryan265472-blip, 2026-01-06:
5in not displaying GPU info