mathoudebine/turing-smart-screen-python #550: WSL and smart screen support

wwdaher, 2024-08-07:
**Describe the bug**  
Running the system monitor under WSL results in [ERROR] Cannot open COM port /dev/ttyACM0

When running main.py under WSL:
line 189 of main.py calls:     display.initialize_display()
The first thing this routine does is call self.lcd.Reset().
This shouldn't be a problem, but under WSL this disconnects the screen device from the subsystem. Unfortunately, it must then be manually reconnected from outside the subsystem via the "usbipd attach --wsl --busid x-x" command before continuing.

Commenting out the self.lcd.Reset() allows it to work as expected.

**To Reproduce**  
Steps to reproduce the behavior:
1. Install the package into a WSL
2. Connect the Turing screen
3. Attach the screen to the WSL USB subsystem with "usbipd attach --wsl --busid x-x" command under powershell
4. Run "python3 configure.py" to configure the screen
5. Run "python3 main.py"

**Expected behavior**  
The screen becomes active and displays system activity.

**Screenshots / photos of the Turing screen**  

**Environment:**  
 - Turing 3.5"
 - Revision of this project [3.5.0, `main` branch]
 - OS with version (Ubuntu 22.04.4 LTS installed under Windows 11 WSL) 
 - Python version [Python 3.10.12]

**Additional context**  
As an aside, if debugging with VSCode under WSL the res directory in configure.py is not properly resolved and has to be coded to be fully qualified (i.e. themes_dir = '/home/thesuser/git/turing-smart-screen-python/res/themes').

mathoudebine, 2024-09-09:
Hi, thanks for the feedback. You may be the first person to run this project on WSL
Out of curiosity, does it run well? Do you have access to all hardware sensors?

wwdaher, 2024-09-09:
Hello Mathieu,

It seems to run fine, though I’ve unfortunately not had time recently to play with it much more. I have it working and drawing various things, but from now I’ll likely be developing on a Raspberry Pi as the dev environment is more integrated than WSL.

There must be a way to get the screen to react faster than it currently does. Is there no hardware assist onboard? I’d have thought there should be a way to send drawing commands directly to the inbuilt processor – or at least load a small kernel that will do the essential drawing operations (like the original MacOS – 90k of tight machine code).

I’ve also tried upping the serial port speed but it hasn’t made any difference.

Thanks for your work on this – it’s saved me a lot of time looking at USB traces!

Best regards,
Will

mathoudebine, 2025-10-15:
Closing as the config. option RESET_ON_STARTUP is now available in latest release. Paths have also been reworked

closed 2025-10-15 by mathoudebine