mathoudebine/turing-smart-screen-python #566: Support for AMD GPU and CPU

jpegCristo, 2024-09-04:
Hi, first of all thanks for the software. It works perfectly and I was able to modify it to my liking without any problems.
I tried to search everywhere if it was possible to have a working % usage for AMD cpu and gpu with external python libraries but i didn't find anything. I also tried to circumvent the problem with the help of chatgpt (since i'm not able to work with python) by trying to get the % usage values ​​directly from windows or in any case from some AMD software but there was no solution.

So I ask you, is there a way to solve this problem? Or is a resolution already planned for those who unfortunately use recent AMD cards and processors?

Thanks so much to everyone who responds.

(Owner of 5800 X3D and RX 7900 GRE)

c0ldJS, 2024-09-07:
AMD CPUs are supported, and GPUs have some support but there are platform and resource factors. You've consulted the table and notes on [this page](https://github.com/mathoudebine/turing-smart-screen-python/wiki/System-monitor-:-hardware-support)?

The box I'm monitoring is Ryzen flavored; theme displays utilization and temperature for the processor.

jpegCristo, 2024-09-07:
@c0ldJS yes, i've already consulted the table you linked but i still have some issues with % usage for both cpu and gpu.

c0ldJS, 2024-09-07:
If this all pertains to a Windows flavored box then recheck the linked page because there's a note that singles out AMD GPU monitoring, where the obstacles are, and how to move forward in the interim. Some of the things getting in the way are upstream to this project.

How do you have the `Hardware Monitoring` option set in `configure.py`, and do issues with displaying a percentage for CPU persist when this is changed (even to the faked data settings)? When moving to the LibreHardwareMonitor selection, recall that you may need to [kick Windows in the teeth](https://github.com/mathoudebine/turing-smart-screen-python/wiki/Troubleshooting#windows-cannot-load-librehardwaremonitordll-loadfromremotesources) first.

Is this happening on one of the stock themes supplied with the project? Or one you're making yourself?