mathoudebine/turing-smart-screen-python #590: tkinter error on windows 11

figurenine, 2024-10-20:
Tkinter appears to be installed correctly, but I can't run

`python.exe .\configure.py`

[image]
[image]
[image]

mathoudebine, 2024-10-30:
Hi @figurenine
Tkinter seems to be installed, I think the error message is erroneous and indicate an issue with another dependency.
I see you are using Python 13. I did not check the compatibility of this project with Python 13 for now so it is recommended to use Python 12 that you can download here [python.org/downloads/release/python-3127](https://www.python.org/downloads/release/python-3127/)

AskoPuu, 2024-11-10:
I have the same problem, i tried with Python 3.12.7. I confirmed Tinker is installed.

mathoudebine, 2024-11-13:
Can you try this change in `configure.py`?
Please replace these lines:
https://github.com/mathoudebine/turing-smart-screen-python/blob/4ca1f5ffe49dc3deafee210498294749f95d19ec/configure.py#L36-L46
by:
```python
 import tkinter.ttk as ttk 
 from tkinter import * 
 from PIL import ImageTk 
```
(be careful there is one blank space at the beginning of each line)
It should display a different error message that you can then post here

AskoPuu, 2024-11-15:
Now it actually loads the configure window (didn't previously).

Windows is saying (to my understanding) that the files are untrusted. I unblocked most of the libre-files, and it continue but nothing shows in the screen.

C:\Users\Administrator\Documents\turing-smart-screen-python-3.5.0>15.11.2024 14.16.44 [INFO] Loading theme 5inchTheme2Radial from res/themes/5inchTheme2Radial/theme.yaml
15.11.2024 14.16.44 [DEBUG] HW revision: C
15.11.2024 14.16.44 [DEBUG] Static COM port: COM3
System.NotSupportedException: An attempt was made to load an assembly from a network location which would have caused the assembly to be sandboxed in previous versions of the .NET Framework. This release of the .NET Framework does not enable CAS policy by default, so this load may be dangerous. If this load is not intended to sandbox the assembly, please enable the loadFromRemoteSources switch. See http://go.microsoft.com/fwlink/?LinkId=155569 for more information.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\Administrator\Documents\turing-smart-screen-python-3.5.0\main.py", line 64, in <module>
    import library.scheduler as scheduler
  File "C:\Users\Administrator\Documents\turing-smart-screen-python-3.5.0\library\scheduler.py", line 28, in <module>
    import library.stats as stats
  File "C:\Users\Administrator\Documents\turing-smart-screen-python-3.5.0\library\stats.py", line 67, in <module>
    import library.sensors.sensors_librehardwaremonitor as sensors
  File "C:\Users\Administrator\Documents\turing-smart-screen-python-3.5.0\library\sensors\sensors_librehardwaremonitor.py", line 40, in <module>
    clr.AddReference(lhm_dll)
System.IO.FileLoadException: Could not load file or assembly 'file:///C:\Users\Administrator\Documents\turing-smart-screen-python-3.5.0\external\LibreHardwareMonitor\LibreHardwareMonitorLib.dll' or one of its dependencies. Operation is not supported. (Exception from HRESULT: 0x80131515)
File name: 'file:///C:\Users\Administrator\Documents\turing-smart-screen-python-3.5.0\external\LibreHardwareMonitor\LibreHardwareMonitorLib.dll' ---> System.NotSupportedException: An attempt was made to load an assembly from a network location which would have caused the assembly to be sandboxed in previous versions of the .NET Framework. This release of the .NET Framework does not enable CAS policy by default, so this load may be dangerous. If this load is not intended to sandbox the assembly, please enable the loadFromRemoteSources switch. See http://go.microsoft.com/fwlink/?LinkId=155569 for more information.
   at System.Reflection.RuntimeAssembly._nLoad(AssemblyName fileName, String codeBase, Evidence assemblySecurity, RuntimeAssembly locationHint, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   at System.Reflection.RuntimeAssembly.InternalLoadAssemblyName(AssemblyName assemblyRef, Evidence assemblySecurity, RuntimeAssembly reqAssembly, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntrospection, Boolean suppressSecurityChecks)
   at System.Reflection.RuntimeAssembly.InternalLoadFrom(String assemblyFile, Evidence securityEvidence, Byte[] hashValue, AssemblyHashAlgorithm hashAlgorithm, Boolean forIntrospection, Boolean suppressSecurityChecks, StackCrawlMark& stackMark)
   at System.Reflection.Assembly.LoadFrom(String assemblyFile)
   at Python.Runtime.AssemblyManager.LoadAssemblyFullPath(String name)
   at Python.Runtime.CLRModule.AddReference(String name)

AoXNeR, 2024-12-10:
> Can you try this change in `configure.py`? Please replace these lines: [...]

hi,
tried this edit,
new error is: ModuleNotFoundError: No module named 'PIL'
after installing Pillow, the new error is:
[ERROR] Python dependencies not installed.

SOLUTION:
-Install Python 12 from your link
-Install Pillow (python3 -m pip install Pillow)
-Install dependencies (tutorial)
-Run configure.py

Fixed the issues for me.

mathoudebine, 2024-12-22:
This issue should be resolved with the latest commits from the 'main' branch, please tag me here if it is not

closed 2024-12-22 by mathoudebine