mathoudebine/turing-smart-screen-python #591: Resume from hibernation (W10, task scheduler, screen is corrupted)

Memosis, 2024-10-23:
I run main.py from Task Scheduler on my Windows 10 machine for Turing 3.5" model in landscape mode . It works nicely after clean system start, it also goes nicely to black screen when I hibernate. Only after resuming from hibernation the screen turns on with corrupted image, it's not readable, some pixels are flashing. I need to click on systray and select Configure (screen goes black) and "Save and run" to restore the functionality (or disconnect and reconnect the USB cable and run scheduler task again). Is there something I can configure differently to deal with this hibernation issue or is that something that's not implemented yet?

macfly92, 2024-12-19:
Hi ! I have the same issue. Worked well since 1 year, and begin to do this behaviour just after a Windows update or Bios update (was applied both same time).
@Memosis did you found a workaround for this ?
Thanks.

Memosis, 2024-12-29:
I have one more observation. I usually turn off the power switch to my PC and monitors after I hibernate. And that leads to the corruption of the screen as described in the issue.

But when I LEAVE the power switch on (PC hibernated (with USB ports still powered?), monitors on standby), it works without any corruption after waking up from hibernation.

Villanut72, 2025-01-23:
I have the same bug even with the power on everytime the pc sleeps the screen is corrupted.

macfly92, 2025-05-14:
Hi !
Finally got time to dig around and was able to fix it.
I had to (re)enable the "USB Standby Power at S4/S5" in the power management of my motherboard's BIOS. 

[image]

Hope this helps.