mathoudebine/turing-smart-screen-python #592: GPU Stats not updating

elschnitzel, 2024-10-23:
**Describe the bug**  
Regardless of the Theme and if I use LibreHardwareMonitor or not. My GPU Stats (percentage and temp) are not updating in the specified interval. All other stats seem to be updating correctly.

**To Reproduce**  
Does not matter if I start with scheduled task and pyhtonw.exe or in console behaviour is always like that.

**Expected behavior**  
GPU Stats update according to the set interval in the theme.

**Screenshots / photos of the Turing screen**  

Log File:
23/10/2024 21:18:16 [INFO] Loading theme Fallout from res/themes/Fallout/theme.yaml
23/10/2024 21:18:16 [DEBUG] HW revision: A
23/10/2024 21:18:16 [DEBUG] Auto detected COM port: COM3
23/10/2024 21:18:16 [DEBUG] Found LibreHardwareMonitorLib 0.9.3.0
23/10/2024 21:18:18 [INFO] Found CPU: 13th Gen Intel Core i7-13700K
23/10/2024 21:18:18 [INFO] Found Memory: Generic Memory
23/10/2024 21:18:18 [INFO] Found Nvidia GPU: NVIDIA GeForce RTX 3070
23/10/2024 21:18:18 [INFO] Found Intel GPU: Intel(R) UHD Graphics 770
23/10/2024 21:18:18 [INFO] Found Storage: CT2000P3SSD8
23/10/2024 21:18:18 [INFO] Found Storage: Crucial_CT525MX300SSD1
23/10/2024 21:18:18 [INFO] Found Storage: TS2TMTE220S
23/10/2024 21:18:18 [INFO] Found Network interface: Bluetooth Network Connection 3
23/10/2024 21:18:18 [INFO] Found Network interface: Ethernet 2
23/10/2024 21:18:18 [DEBUG] Using Python 3.12.7 (tags/v3.12.7:0b05ead, Oct  1 2024, 03:06:41) [MSC v.1941 64 bit (AMD64)]
23/10/2024 21:18:18 [INFO] Tray icon has been displayed
23/10/2024 21:18:18 [INFO] Display reset (COM port may change)...
23/10/2024 21:18:23 [DEBUG] Static COM port: COM3
23/10/2024 21:18:24 [DEBUG] HW sub-revision: SubRevision.TURING_3_5
23/10/2024 21:18:24 [DEBUG] Drawing Image: BACKGROUND
23/10/2024 21:18:24 [DEBUG] Bitmap res/themes/Fallout/background_fo.png is now loaded in the cache
23/10/2024 21:18:24 [DEBUG] Drawing Text: TEXT_CPU_
23/10/2024 21:18:24 [DEBUG] Drawing Text: TEXT_CPU
23/10/2024 21:18:24 [DEBUG] Drawing Text: TEXT_GPU_
23/10/2024 21:18:24 [DEBUG] Drawing Text: TEXT_GPU
23/10/2024 21:18:24 [DEBUG] Drawing Text: TEXT_RAM_FREE
23/10/2024 21:18:24 [DEBUG] Drawing Text: TEXT_RAM_USED
23/10/2024 21:18:24 [DEBUG] Drawing Text: TEXT_RAM_LOAD
23/10/2024 21:18:24 [DEBUG] Drawing Text: TEXT_HDD_PERCENT
23/10/2024 21:18:24 [DEBUG] Drawing Text: TEXT_HDD_FREE
23/10/2024 21:18:24 [DEBUG] Drawing Text: TEXT_HDD_TOTAL
23/10/2024 21:18:24 [WARNING] Found 2 GPUs on your system (0 AMD / 1 Nvidia / 1 Intel). Auto identify which GPU to use.
23/10/2024 21:18:24 [DEBUG] This GPU will be used for stats: NVIDIA GeForce RTX 3070

**Environment:**  
 - Smart screen model: 3.5 inch turing
 - Revision of this project 3.5.0
 - OS with version  Windows 11
 - Python version  Python 3.12
 - Hardware Intel 13700k, Nvidia 3070

**Additional context**  
Add any other context about the problem here.

MAHAON26, 2024-10-28:
Having same issue, I got only one GPU, it detects normally. Screen shows starting data and freeze all info about GPU.

MAHAON26, 2024-10-30:
I did exactly as in your correct example, gpu stats now working, but cpu stats start freezing as gpu was. Can I share my theme and log so you can help us?

MAHAON26, 2024-10-30:
My Theme (will share it when done):
[My Custom Theme.zip](https://github.com/user-attachments/files/17575684/My.Custom.Theme.zip)

My Log:

30.10.2024 4:58:19 PM [INFO] Loading theme My Custom Theme from res/themes/My Custom Theme/theme.yaml
30.10.2024 4:58:20 PM [DEBUG] HW revision: A
30.10.2024 4:58:20 PM [DEBUG] Auto detected COM port: COM8
30.10.2024 4:58:21 PM [DEBUG] Found LibreHardwareMonitorLib 0.9.3.0
30.10.2024 4:58:23 PM [INFO] Found CPU: AMD Ryzen 5 2600
30.10.2024 4:58:23 PM [INFO] Found Memory: Generic Memory
30.10.2024 4:58:23 PM [INFO] Found AMD GPU: AMD Radeon RX 5600 XT
30.10.2024 4:58:23 PM [INFO] Found Storage: KINGSTON SA2000M81000G
30.10.2024 4:58:23 PM [INFO] Found Storage: ADATA SX6000PNP
30.10.2024 4:58:23 PM [INFO] Found Network interface: Ethernet
30.10.2024 4:58:23 PM [INFO] Found Network interface: Сетевое подключение Bluetooth
30.10.2024 4:58:23 PM [DEBUG] Using Python 3.10.11 (tags/v3.10.11:7d4cc5a, Apr  5 2023, 00:38:17) [MSC v.1929 64 bit (AMD64)]
30.10.2024 4:58:23 PM [INFO] Tray icon has been displayed
30.10.2024 4:58:23 PM [INFO] Display reset (COM port may change)...
30.10.2024 4:58:28 PM [DEBUG] Static COM port: COM8
30.10.2024 4:58:30 PM [DEBUG] HW sub-revision: SubRevision.TURING_3_5
30.10.2024 4:58:30 PM [DEBUG] Drawing Image: BACKGROUND
30.10.2024 4:58:30 PM [DEBUG] Bitmap res/themes/My Custom Theme/background.png is now loaded in the cache
30.10.2024 4:58:30 PM [DEBUG] Drawing Text: CPU_MODEL
30.10.2024 4:58:30 PM [DEBUG] Drawing Text: GPU_MODEL
30.10.2024 4:58:30 PM [DEBUG] Drawing Text: CPU_TEXT
30.10.2024 4:58:30 PM [DEBUG] Drawing Text: CPU_FREQ
30.10.2024 4:58:30 PM [DEBUG] Drawing Text: GPU_TEXT
30.10.2024 4:58:30 PM [DEBUG] Drawing Text: GPU_FREQ
30.10.2024 4:58:30 PM [DEBUG] Drawing Text: GPU_MEM
30.10.2024 4:58:30 PM [DEBUG] Drawing Text: GPU_FPS
30.10.2024 4:58:30 PM [DEBUG] Drawing Text: MEM_USED
30.10.2024 4:58:30 PM [DEBUG] Drawing Text: MEM_FREE
30.10.2024 4:58:30 PM [DEBUG] Found one supported GPU: AMD Radeon RX 5600 XT
30.10.2024 4:58:30 PM [WARNING] Your GPU memory relative usage (%) is not supported yet

Thx for your help in advance!)

MAHAON26, 2024-10-30:
Also, I use app named Fan Control, it also use LibreHardwareMonitor. Can they somehow conflict? I already try to disable it and it no help with freezing stats

elschnitzel, 2024-11-01:
Thank you @mathoudebine,

That was exactly what caused the issue,

@MAHAON26 The solution is only for the GPU stats. CPU stats still need to have their own interval.

Your Code:
```yaml
  CPU:
    INTERVAL: 1
    PERCENTAGE:
      TEXT:
        SHOW: True
        SHOW_UNIT: True
        X: 0
        Y: 60
        FONT: generale-mono/GeneraleMonoA.ttf
        FONT_SIZE: 40
        FONT_COLOR: 2, 216, 243
        BACKGROUND_IMAGE: background.png
      GRAPH:
        SHOW: True
        X: 10
        Y: 110
        WIDTH: 101
        HEIGHT: 13
        MIN_VALUE: 0
        MAX_VALUE: 100
        BAR_COLOR: 2, 216, 243
        BAR_OUTLINE: True
        BACKGROUND_IMAGE: background.png
    TEMPERATURE:
      TEXT:
        SHOW: True
        SHOW_UNIT: True
        X: 385
        Y: 45
        FONT: generale-mono/GeneraleMonoA.ttf
        FONT_SIZE: 20
        FONT_COLOR: 2, 216, 243
        BACKGROUND_IMAGE: background.png 
    FREQUENCY:
      TEXT:
        SHOW: True
        SHOW_UNIT: True
        X: 127.5
        Y: 70
        FONT: generale-mono/GeneraleMonoA.ttf
        FONT_SIZE: 20
        FONT_COLOR: 2, 216, 243
        BACKGROUND_IMAGE: background.png
      GRAPH:
        SHOW: True
        X: 130
        Y: 110
        WIDTH: 101
        HEIGHT: 13
        MIN_VALUE: 0
        MAX_VALUE: 4
        BAR_COLOR: 2, 216, 243
        BAR_OUTLINE: True
        BACKGROUND_IMAGE: background.png
```
        
Should work like that:
```yaml
  CPU:
    PERCENTAGE:
      INTERVAL: 1
      TEXT:
        SHOW: True
        SHOW_UNIT: True
        X: 0
        Y: 60
        FONT: generale-mono/GeneraleMonoA.ttf
        FONT_SIZE: 40
        FONT_COLOR: 2, 216, 243
        BACKGROUND_IMAGE: background.png
      GRAPH:
        SHOW: True
        X: 10
        Y: 110
        WIDTH: 101
        HEIGHT: 13
        MIN_VALUE: 0
        MAX_VALUE: 100
        BAR_COLOR: 2, 216, 243
        BAR_OUTLINE: True
        BACKGROUND_IMAGE: background.png
    TEMPERATURE:
      INTERVAL: 1
      TEXT:
        SHOW: True
        SHOW_UNIT: True
        X: 385
        Y: 45
        FONT: generale-mono/GeneraleMonoA.ttf
        FONT_SIZE: 20
        FONT_COLOR: 2, 216, 243
        BACKGROUND_IMAGE: background.png 
    FREQUENCY:
      INTERVAL: 1
      TEXT:
        SHOW: True
        SHOW_UNIT: True
        X: 127.5
        Y: 70
        FONT: generale-mono/GeneraleMonoA.ttf
        FONT_SIZE: 20
        FONT_COLOR: 2, 216, 243
        BACKGROUND_IMAGE: background.png
      GRAPH:
        SHOW: True
        X: 130
        Y: 110
        WIDTH: 101
        HEIGHT: 13
        MIN_VALUE: 0
        MAX_VALUE: 4
        BAR_COLOR: 2, 216, 243
        BAR_OUTLINE: True
        BACKGROUND_IMAGE: background.png
```

closed 2024-11-01 by elschnitzel