mathoudebine/turing-smart-screen-python #593: Getting requirements to build hweel did not run unsccessfully.

tim5686, 2024-10-28:
**Describe the bug**  
When I try to add the python dependencies, I am gettin gan error that the subprocess exited with error. Getting requirements to build wheel did not run successfully

**To Reproduce**  
When  running python.exe .\configure.py - i run into the error. It says tkinter not installed, but following the help page, I can confirm that the option is ticked on installation

**Expected behavior**  
A clear and concise description of what you expected to happen.

**Screenshots / photos of the Turing screen**  
This is what happens when I run the requirements.txt dependencies:
C:\Users\revti\Downloads\turing-smart-screen-python-3.5.0\turing-smart-screen-python-3.5.0>python.exe -m pip install -r requirements.txt
Ignoring numpy: markers 'python_version < "3.9"' don't match your environment
Ignoring GPUtil: markers 'python_version < "3.12"' don't match your environment
Collecting GPUtil@ git+https://github.com/mathoudebine/gputil.git@1.4.0-py3.12 (from -r requirements.txt (line 19))
  Cloning https://github.com/mathoudebine/gputil.git (to revision 1.4.0-py3.12) to c:\users\revti\appdata\local\temp\pip-install-sy6hra4p\gputil_cd41900e6a27439d8d55d133acc2e26d
  Running command git clone --filter=blob:none --quiet https://github.com/mathoudebine/gputil.git 'C:\Users\revti\AppData\Local\Temp\pip-install-sy6hra4p\gputil_cd41900e6a27439d8d55d133acc2e26d'
  Resolved https://github.com/mathoudebine/gputil.git to commit 1c6e890a5d9592d69ffcf50d31f12e89699b8c8e
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Ignoring pyamdgpuinfo: markers 'sys_platform == "linux" and python_version <= "3.12"' don't match your environment
Collecting Pillow~=10.3.0 (from -r requirements.txt (line 2))
  Using cached pillow-10.3.0.tar.gz (46.6 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [21 lines of output]
      Traceback (most recent call last):
        File "C:\Users\revti\AppData\Local\Programs\Python\Python313\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 353, in <module>
          main()
          ~~~~^^
        File "C:\Users\revti\AppData\Local\Programs\Python\Python313\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ~~~~^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\revti\AppData\Local\Programs\Python\Python313\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
        File "C:\Users\revti\AppData\Local\Temp\pip-build-env-aks2b7g3\overlay\Lib\site-packages\setuptools\build_meta.py", line 332, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=[])
                 ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\revti\AppData\Local\Temp\pip-build-env-aks2b7g3\overlay\Lib\site-packages\setuptools\build_meta.py", line 302, in _get_build_requires
          self.run_setup()
          ~~~~~~~~~~~~~~^^
        File "C:\Users\revti\AppData\Local\Temp\pip-build-env-aks2b7g3\overlay\Lib\site-packages\setuptools\build_meta.py", line 318, in run_setup
          exec(code, locals())
          ~~~~^^^^^^^^^^^^^^^^
        File "<string>", line 33, in <module>
        File "<string>", line 27, in get_version
      KeyError: '__version__'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

**Environment:**  
 - Smart screen model [Turing 3.5]
 - Revision of this project [3.5.0]
 - OS with version [Windows 11]
 - Python version [Python 3.13.0]
 - Hardware [Intel CPU (i9-12900k, Nvidia GPU RTX 3080]

**Additional context**  
Add any other context about the problem here.

mathoudebine, 2024-10-30:
Hi @tim5686 
I did not check the compatibility of this project with Python 13 for now. Looking at your logs it seems one of the dependency (GPutil) is not fully compatible with Python 13 yet.
Until this is fixed, I recommend to use Python 12 that you can download here https://www.python.org/downloads/release/python-3127/

tim5686, 2024-11-02:
Thanks @mathoudebine all working now!

closed 2024-11-02 by tim5686