mathoudebine/turing-smart-screen-python #605: FileNotFoundError: [Errno 2] No such file or directory: 'config.yaml'

allexmail, 2024-11-13:
When trying to start main.py not from the installation directory, an error occurs "FileNotFoundError: [Errno 2] No such file or directory: 'config.yaml'"

**To Reproduce**  
1. Install the python virtual environment in /opt/USBmon/
2. Download the source codes and unzip them into a folder /opt/USBMon/turing-smart-screen-3.5.0/
3. Install the necessary libraries in the virtual environment.
4. Make a configuration of your screen and save it.
5. Run the main.py script and make sure everything is working: /opt/USBMon/bin/python3 /opt/USBMon/turing-smart-screen-3.5.0/main.py (Screenshot1)
6. Open new terminal. 
7. Try to run the main script from the home directory: /opt/USBMon/bin/python3 /opt/USBMon/turing-smart-screen-3.5.0/main.py (Screenshot2)

**Expected behavior**  
I expected to see the theme display on the USB screen.

**Screenshots / photos of the Turing screen**  
[image]
[image]

**Environment:**  
 - Smart screen model XuanFang 3.5, COM port: /dev/ttyACM0
 - Revision of this project 3.0.5, main
 - OS with version Ubuntu 24.04
 - Python version Python 3.12.3 (main, Sep 11 2024, 14:17:37) [GCC 13.2.0]
 - Hardware Intel CPU

**Additional context**  
There has already been such a [problem](https://github.com/mathoudebine/turing-smart-screen-python/issues/150), but it has been closed without a solution. My task is to launch the screen from the startup script, which, of course, is not in the installation directory of your program. Let me know if you need more information.

mathoudebine, 2024-12-22:
Hi @allexmail 
It is a known issue that I didn't have time to fix, you can check #358
For now you have to run `main.py` inside the code folder because the `config.yaml` is seeked inside the current folder
I hope to fix it in a future version

mathoudebine, 2024-12-22:
Hi again @allexmail 
The issue should now be fixed in the main branch, don't hesitate to tag me here if it is not the case

closed 2024-12-22 by mathoudebine