mathoudebine/turing-smart-screen-python #616: TSSP 3.5.0 doesn't start on Arch Linux system; NumPy incompatibility

emk2203, 2024-12-13:
**Describe the bug**  
A clear and concise description of what the bug is.

**To Reproduce**  
Steps to reproduce the behavior:
1. I installed v3.5.0 in `/opt/turing-smart-screen-python` and edited `config.yaml` according to my configuration. Installed Babel and NumPy with `pacman`, commented GPUtil in `sensors_python.py` since it's not needed (hardware is Intel NUC6CAYH, only internal Intel GPU)
2. Since this is running on a bare-bones system, no clicks and scrolls; everything done on command line.
3. User is part of `uucp` group and the permissions are set to allow write access for user in program directory (owner root, group uucp, permissions 0664/0775)
4. Starting program with `systemctl`. service file resides in `/usr/local/lib/systemd/system/`

**Expected behavior**  
Program starts up and drives the display, `systemctl status ...` without errors

**Screenshots / photos of the Turing screen**  
Nothing happens; log file (`journalctl --unit=turing-smart-screen-python.service --since "1 hour ago"`)

```
Dec 13 16:15:52 NUC6CAYH systemd[1]: Started Turing Smart Screen.
Dec 13 16:15:53 NUC6CAYH python3[741]: 12/13/2024 04:15:53 PM [INFO] Loading theme 3.5inchTheme2 from res/themes/3.5inchTheme2/theme.yaml
Dec 13 16:15:53 NUC6CAYH python3[741]: 12/13/2024 04:15:53 PM [DEBUG] HW revision: A
Dec 13 16:15:53 NUC6CAYH python3[741]: 12/13/2024 04:15:53 PM [DEBUG] Auto detected COM port: /dev/ttyACM0
Dec 13 16:15:53 NUC6CAYH python3[741]: 12/13/2024 04:15:53 PM [DEBUG] Using Python 3.12.7 (main, Oct  1 2024, 11:15:50) [GCC 14.2.1 20240910]
Dec 13 16:15:53 NUC6CAYH python3[741]: 12/13/2024 04:15:53 PM [WARNING] Tray icon is not supported on your platform
Dec 13 16:15:53 NUC6CAYH python3[741]: 12/13/2024 04:15:53 PM [INFO] Display reset (COM port may change)...
Dec 13 16:15:58 NUC6CAYH python3[741]: 12/13/2024 04:15:58 PM [DEBUG] Static COM port: /dev/ttyACM0
Dec 13 16:15:59 NUC6CAYH python3[741]: 12/13/2024 04:15:59 PM [DEBUG] HW sub-revision: SubRevision.TURING_3_5
Dec 13 16:15:59 NUC6CAYH python3[741]: 12/13/2024 04:15:59 PM [DEBUG] Drawing Image: BACKGROUND
Dec 13 16:15:59 NUC6CAYH python3[741]: 12/13/2024 04:15:59 PM [DEBUG] Bitmap res/themes/3.5inchTheme2/background.png is now loaded in the cache
Dec 13 16:15:59 NUC6CAYH python3[741]: Traceback (most recent call last):
Dec 13 16:15:59 NUC6CAYH python3[741]:   File "/opt/turing-smart-screen-python-3.5.0/main.py", line 192, in <module>
Dec 13 16:15:59 NUC6CAYH python3[741]:     display.display_static_images()
Dec 13 16:15:59 NUC6CAYH python3[741]:   File "/opt/turing-smart-screen-python-3.5.0/library/display.py", line 113, in display_static_images
Dec 13 16:15:59 NUC6CAYH python3[741]:     self.lcd.DisplayBitmap(
Dec 13 16:15:59 NUC6CAYH python3[741]:   File "/opt/turing-smart-screen-python-3.5.0/library/lcd/lcd_comm.py", line 205, in DisplayBitmap
Dec 13 16:15:59 NUC6CAYH python3[741]:     self.DisplayPILImage(image, x, y, width, height)
Dec 13 16:15:59 NUC6CAYH python3[741]:   File "/opt/turing-smart-screen-python-3.5.0/library/lcd/lcd_comm_rev_a.py", line 234, in DisplayPILImage
Dec 13 16:15:59 NUC6CAYH python3[741]:     rgb565le = self.imageToRGB565LE(image)
Dec 13 16:15:59 NUC6CAYH python3[741]:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
Dec 13 16:15:59 NUC6CAYH python3[741]:   File "/opt/turing-smart-screen-python-3.5.0/library/lcd/lcd_comm_rev_a.py", line 199, in imageToRGB565LE
Dec 13 16:15:59 NUC6CAYH python3[741]:     return rgb565.newbyteorder('<').tobytes()
Dec 13 16:15:59 NUC6CAYH python3[741]:            ^^^^^^^^^^^^^^^^^^^
Dec 13 16:15:59 NUC6CAYH python3[741]: AttributeError: `newbyteorder` was removed from the ndarray class in NumPy 2.0. Use `arr.view(arr.dtype.newbyteorder(order))` inst>
Dec 13 16:15:59 NUC6CAYH python3[741]: 12/13/2024 04:15:59 PM [INFO] Program will now exit
Dec 13 16:15:59 NUC6CAYH python3[741]: 12/13/2024 04:15:59 PM [INFO] Waiting for all pending request to be sent to display (5s max)...
```

There seems to  be an incompatibility with NumPy 2.0 according to the last lines.

**Environment:**  
 - Smart screen model [Turing 3.5”]
 - Revision of this project [3.5.0]
 - OS with version [Arch Linux 2024-12-13]
 - Python version [Python 3.12]
 - Hardware [Intel CPU J3455 @ 1.50GHz with i915 GPU]

**Additional context**  
Add any other context about the problem here.

mathoudebine, 2024-12-21:
Hi @emk2203 
Numpy 2.x support is now implemented on main branch and will be included on the next release

closed 2024-12-21 by mathoudebine

emk2203, 2024-12-22:
Would it work to clone main, or is it wip and not advised yet?

mathoudebine, 2024-12-22:
Main branch should be fully functional at any time, you can use it if you like. I create releases for people that do not use Git, as a way to store changelogs and warn for breaking changes (if any)