mathoudebine/turing-smart-screen-python #625: Issue with Task Scheduler

ispascelgras, 2024-12-23:
Hey, sorry for asking.

I have an issue with the Task Scheduler and I was wondering if you have any ideas of how to fix it.
I follow all the steps, but when I check "Run with highest privileges" and press OK I receive the following error.

[image]

Everything is running on Administrator rights, I've already performed the sfc /scannow.

Thanks!

c0ldJS, 2024-12-24:
This is an operating system matter, not a bug with `turing-smart-screen-python`, and should probably move to Discussions.

Fully scheduling the task requires configurations on the `General`, `Triggers`, and `Actions` tabs and missing any of them can produce that error. Additionally, if (for some strange reason) Task Manager isn't participating in the `SYSTEM` group then the security options go sideways and that error can also be returned.

In the provided screenshot, is the `Details` column showing the full path you configured for an action or is that being cut off?

ispascelgras, 2024-12-25:
Hey, 

Thanks for answering! 
Task Manager runs as SYSTEM, which is in the same group as my user "iSPAS"
I'm publishing some screenshots for context:

[image]
[image]
[image]
[image]

_In the provided screenshot, is the Details column showing the full path you configured for an action or is that being cut off?_

Yes, it's the full path to `pythonw.exe`, as shown in the 3rd screenshot. The Start in goes to the Root folder of `\turing-smart-screen-python-3.6.0\turing-smart-screen-python-3.6.0`

I have to tell you that I run on Windows 11.

Is there anything else you need, or does it ring a bell from what I already posted?

Thanks and Merry Christmas :)!

AoXNeR, 2024-12-25:
same exact issue here.

c0ldJS, 2024-12-25:
In that last screenshot, when `Start in:` is being configured, is that a full path including the relevant drive? Or is it precisely `\turing-smart-screen-python-3.6.0\turing-smart-screen-python-3.6.0`?

AoXNeR, 2024-12-26:
Fixed when changing task user to "Administrators "

ispascelgras, 2024-12-26:
Thank you all for your help!, @AoXNeR was right, that fixed the issue.

mathoudebine, 2024-12-27:
Thanks, I updated the wiki page with the fix!

closed 2024-12-27 by mathoudebine